RobotLocomotion/drake #12546: need animate/recording feature in planar_scenegraph_visualizer for jupyter integration

RussTedrake, 2020-01-08:
This was discussed during the initial PR for planar_scenegraph_visualizer.  Now it has reared it's head -- I cannot yet replace my underactuated version of `planar_scenegraph_visualizer` with the drake version because I don't know how to get the jupyter workflow yet.
https://github.com/RussTedrake/underactuated/pull/244
The goal is to be able to have the pyplot output of planar_scenegraph_visualizer appear as an animation in the juypter console (as opposed to a separate window).  

Original discussion:
https://github.com/robotlocomotion/drake/pull/12218#issuecomment-550491446

After that thread, it seems that the `animate` was removed, too.  

cc @EricCousineau-TRI in case he has a different trick for integrating the pyplot animation into a jupyter notebook.

RussTedrake, 2020-01-13:
Possible solution: we add a `ValueLogger`system (analogous to `SignalLogger`... which feels now like it should be `VectorLogger`) that can log `AbstractValue` inputs... with the intended use case being geometry poses.  

To be useful, we also want to be able to get the recording back out into a system that acts like a `TrajectorySource` (as seen in https://github.com/RobotLocomotion/drake/blob/master/examples/acrobot/test/run_swing_up_traj_optimization.cc#L84 ).  Perhaps the `ValueLogger` system could just always have both the input and output port, but have a boolean state for whether it is recording or playing back?  (I'm open to suggestions for a better design).

RussTedrake, 2020-01-14:
cc @SeanCurtis-TRI

RussTedrake, 2020-01-16:
Just saw #12582.  Awesome!

Prepping for our f2f, I realized that i’ve slightly mixed two related issues here.  One is having an api/capability for playing back simulation results (in the same process) after the sim completes.  The second is having the ability to visualize ANY pyplot sim graphics ever inside a jupyter notebook (we currently achieve this by using pyplot animation and rendering it to html).

EricCousineau-TRI, 2020-01-16:
My understanding is that the main goal for this issue is animation, I think?
Recording is more or less an implementation detail at present (e.g. if a Jupyter widget could show things asynchronously, then that would satisfy the most immediate need), though it will be a useful feature itself later?

SeanCurtis-TRI, 2020-01-16:
I'd recommend that when #12582 merges we close this issue and open a new issue discussing the jupyter requirements. I must admit to huge mountains of ignorance on the subject.

EricCousineau-TRI, 2020-01-16:
I think the main focus of this issue are effectively the Jupyter requirements?

SeanCurtis-TRI, 2020-01-16:
Fair enough. THen this will continue with that focus.

tehbelinda, 2020-01-16:
Now that Eric has added the Jupyter tutorial and it's part of #12582, is that sufficient for the Jupyter requirements?

EricCousineau-TRI, 2020-01-17:
FYI @RussTedrake Here's the Jupyter notebook in Bel's PR:
https://github.com/RobotLocomotion/drake/blob/4c17d4d32a94fb31c8d34eb34a3076e92e7cce3c/tutorials/pyplot_animation_multibody_plant.ipynb

RussTedrake, 2020-01-18:
Yes -- I think #12582 is sufficient for all jupyter requirements that I had requested here... thanks!  eager for it to land!

closed 2020-01-18 by RussTedrake