RobotLocomotion/drake #12548: Pydrake Mathematical Program: Linear Matrix Constraints

sadraddini, 2020-01-08:
Per the discussion I had with @RussTedrake:

I often encounter mathematical programs that have a lot of instances of the following linear constraint: AX~B, where ~ is <=, >=, or ==, A is m*n matrix, X is n*p matrix of variables, and B is m*p matrix. 

Currently drake **AddLinearConstraint** supports Ax~b, where x is a n*1 vector. In order for me to use it for matrices, I have 3 options using **AddLinearConstraint**:

1- Write a _for loop_ for columns (slow for my applications). 
2- Reshape X into a vector and restructure A to A_new accordingly: A_new becomes a very big matrix that is sparse. However, currently drake mathematical program does not support sparse matrices - e.g, it is marked as to do [here](https://github.com/RobotLocomotion/drake/blob/master/solvers/gurobi_solver.cc) for gurobi. Therefore, this option is both slow and not clean. 
3- Write np.equal(np.dot(A,X),B)), and feed it as a vector of formulae to addLinearConstraint. This is often the slowest of all.

I would like either an existing fast solution, or to have the c++ implementation of AX~B with a python binding.

hongkai-dai, 2020-01-08:
Do you already have the `A` and `B` matrix? If so, we can add these two functions
```cpp
Binding<LinearConstraint> MathematicalProgram::AddLinearConstraint(const Eigen::MatrixXd& A, const Eigen::MatrixXd& B, const MatrixX<symbolic::Variable>& X);
Binding<LinearEqualityConstraint> MathematicalProgram::AddLinearEqualityConstraint(const Eigen::MatrixXd& A, const Eigen::MatrixXd& B, const MatrixX<symbolic::Variable>& X);
```

For the return argument of adding AX~B, do you want a single LinearConstraint (this is like option 2) above, or you want a vector of LinearConstraint, with each element in the vector representing A * X.col(i) ~ B.col(i) (like in option 1)?

sadraddini, 2020-01-08:
Yes A, B are given. I would prefer a single linear constraint.

hongkai-dai, 2020-01-08:
OK, then it would be like implementing option 2 in C++. If you think it is slow to do option 2 in python, then I suspect it would also be slow in C++, as they do the same thing (Drake's LinearConstraint currently don't support sparse matrix, so I have to construct a big Eigen::MatrixXd to represent this sparse matrix.) Have you tried to do option 2 in python already? How slow is it?

sadraddini, 2020-01-08:
Seems to be a confusion. Let's not write the sparse matrix - it is way slow (replacing a linear complexity with a quadratic one). Let's just implement the following pseudocode (e.g., for <=):
```
for i=0 to A.rows()
    for j=0 to B.columns()
         A.row(i)*x.row(j)<=B(i,j)
```

hongkai-dai, 2020-01-08:
Why doing it entry-by-entry, instead of column by column? Namely
```
for i in range(X.cols()):
    prog.AddLinearConstraint(A, -np.full((A.rows(),), np.inf), B.col(i), X.col(i))
```
And return a vector of linear constraint, with each element of the vector representing the i'th column of A * X <= B?

sadraddini, 2020-01-08:
That is also possible, albeit I still like a single linear constraint. If I write the for loop above in python, it is slow. Written in C++ and binding with python should be faster (?)

sadraddini, 2020-01-14:
Any progress/misunderstanding/issues-yet-to-be-resolved on this?

hongkai-dai, 2020-01-14:
Oops, sorry, I will work on it tonight.

hongkai-dai, 2020-01-15:
Could you try the branch https://github.com/hongkai-dai/drake/tree/add_linear_matrix_constraint, and see if it improves the speed in python? You could call
```
prog.AddMatrixLinearConstraint(A, LB, UB, X)
```

sadraddini, 2020-01-15:
It slightly improves the speed in python. Nevertheless, it is much cleaner than other methods.
```
Description of methods: 
       Method 1: AddMatrixLinearConstraint (new) 
       Method 2: For-Loop over columns of X with AddLinearConstraint 
       Method 3: Symbolic LB<=AX<=UB 
       Method 4: Vectorized form with sparse A LB<=kron(A,I)X<=UB (not really exploiting sparsity)
*time in seconds
m=10,n=10,p=12
Method 1 = 0.0010600090026855469
Method 2 = 0.0026044845581054688
Method 3 = 0.051264047622680664
Method 4 = 0.0012819766998291016
m=50,n=40,p=60
Method 1 = 0.07201075553894043
Method 2 = 0.0793311595916748
Method 3 = long time
Method 4 = 0.15785741806030273
m=100,n=90,p=20
Method 1 = 0.21079611778259277
Method 2 = 0.23666644096374512
Method 3 = long time
Method 4 = 0.40691399574279785
m=1000,n=3,p=4
Method 1 = 0.13610219955444336
Method 2 = 0.1712789535522461
Method 3 = 1.9271178245544434
Method 4 = 0.26104068756103516
m=10000,n=3,p=4
Method 1 = 12.241854429244995
Method 2 = 12.505882978439331
Method 3 = long time
Method 4 = 79.97047328948975
```

hongkai-dai, 2020-01-15:
Seems that method 2 (for loop each columns) has takes about the same time as method 1? If so, then I am not entirely convinced to add this method. mathematical_program.h is already very big (> 2000 lines), and it is not hard to implement method 2 with a for loop.

sadraddini, 2020-01-16:
Wait I made a mistake last time - I was reporting adding the method 1's time to method 2. It seems there is something unnatural in AddMatrixLinearConstraint that makes it take a long time. Let me share the code:
```
import numpy as np
import time

import pydrake.symbolic as sym
import pydrake.solvers.mathematicalprogram as MP

print("Description of methods: \n \
      Method 1: AddMatrixLinearConstraint (new) \n \
      Method 2: For-Loop over columns of X with AddLinearConstraint \n \
      Method 3: Symbolic LB<=AX<=UB \n \
      Method 4: Vectorized form with sparse A LB<=kron(A,I)X<=UB (not really exploiting sparsity)")
np.random.seed(0)
for m,n,p in [(10,10,12),(50,40,60),(100,20,20),(1000,3,4),(1000,20,10)]:
    prog=MP.MathematicalProgram()
    X=prog.NewContinuousVariables(n,m,"X")
    A=np.random.random((p,n))-0.5
    LB=-np.ones((p,m))*1
    UB=np.ones((p,m))*1
    print("m=%d,n=%d,p=%d"%(m,n,p))
    # 1: Using AddMatrixLinearConstraint
    start_time=time.time()
    prog.AddMatrixLinearConstraint(A, LB, UB, X)
    print("Method 1 =",time.time()-start_time)
    # 2: Using the for loop
    start_time=time.time()
    for column in range(m):
        prog.AddLinearConstraint(A,LB[:,column],UB[:,column],vars=X[:,column])
    print("Method 2 =",time.time()-start_time)
    # 3: Symbolic
    start_time=time.time()
    if m*n*p<2*10**4:
        start_time=time.time()
        c=np.dot(A,X)
        prog.AddLinearConstraint(np.less_equal(c,UB,dtype='object').flatten())
        prog.AddLinearConstraint(np.greater_equal(c,LB,dtype='object').flatten())
        print("Method 3 =",time.time()-start_time)
    else:
        print("Method 3 = long time")
    # 4: Sparse Matrix (not really exploiting sparsity)
    start_time=time.time()
    prog.AddLinearConstraint(np.kron(A,np.eye(m)),LB.flatten(),UB.flatten(),vars=X.flatten())
    print("Method 4 =",time.time()-start_time)
```
And I get the following result on my computer:
```
Description of methods: 
       Method 1: AddMatrixLinearConstraint (new) 
       Method 2: For-Loop over columns of X with AddLinearConstraint 
       Method 3: Symbolic LB<=AX<=UB 
       Method 4: Vectorized form with sparse A LB<=kron(A,I)X<=UB (not really exploiting sparsity)
m=10,n=10,p=12
Method 1 = 0.0004124641418457031
Method 2 = 0.0006234645843505859
Method 3 = 0.035796165466308594
Method 4 = 0.001085042953491211
m=50,n=40,p=60
Method 1 = 0.06661224365234375
Method 2 = 0.008003473281860352
Method 3 = long time
Method 4 = 0.1523420810699463
m=100,n=20,p=20
Method 1 = 0.04013824462890625
Method 2 = 0.008041620254516602
Method 3 = long time
Method 4 = 0.11997056007385254
m=1000,n=3,p=4
Method 1 = 0.130021333694458
Method 2 = 0.03922867774963379
Method 3 = 1.9425733089447021
Method 4 = 0.2644779682159424
m=1000,n=20,p=10
Method 1 = 2.084963798522949
Method 2 = 0.08111786842346191
Method 3 = long time
Method 4 = 4.441803455352783
```

sadraddini, 2020-01-16:
And regarding which one should be the ultimate function: AddLinearConstraint is a subset of AddMatrixLinearConstraint. Why not change AddLinearConstraint such that instead of 
```
self: pydrake.solvers.mathematicalprogram.MathematicalProgram, A: numpy.ndarray[float64[m, n], flags.f_contiguous], lb: numpy.ndarray[float64[m, 1]], ub: numpy.ndarray[float64[m, 1]], vars: numpy.ndarray[object[m, 1]]) -> drake::solvers::Binding<drake::solvers::LinearConstraint>
```
it would be 
```
self: pydrake.solvers.mathematicalprogram.MathematicalProgram, A: numpy.ndarray[float64[m, n], flags.f_contiguous], lb: numpy.ndarray[float64[m, **p**]], ub: numpy.ndarray[float64[m, **p**]], vars: numpy.ndarray[object[m, **p**]]) -> drake::solvers::Binding<drake::solvers::LinearConstraint>
```

hongkai-dai, 2020-01-17:
I changed the implementation of `AddMatrixLinearConstraint`, to use a for loop and add the linear constraint for LB.col(i) <= A * X.col(i) <= UB.col(i). Now this is the new result
```
Description of methods:
       Method 1: AddMatrixLinearConstraint (new)
       Method 2: For-Loop over columns of X with AddLinearConstraint
       Method 3: Symbolic LB<=AX<=UB
       Method 4: Vectorized form with sparse A LB<=kron(A,I)X<=UB (not really exploiting sparsity)
m=10,n=10,p=12
Method 1 = 0.00023746490478515625
Method 2 = 0.00032329559326171875
Method 3 = 0.014430522918701172
Method 4 = 0.0004677772521972656
m=50,n=40,p=60
Method 1 = 0.0019676685333251953
Method 2 = 0.0031898021697998047
Method 3 = long time
Method 4 = 0.12289667129516602
m=100,n=20,p=20
Method 1 = 0.0016803741455078125
Method 2 = 0.0029408931732177734
Method 3 = long time
Method 4 = 0.06867098808288574
m=1000,n=3,p=4
Method 1 = 0.003068208694458008
Method 2 = 0.01084589958190918
Method 3 = 1.065206527709961
Method 4 = 0.20610952377319336
m=1000,n=20,p=10
Method 1 = 0.016391515731811523
Method 2 = 0.027560949325561523
Method 3 = long time
Method 4 = 3.5228700637817383
```
Now method 1 takes less time than method 2, but they both takes a fraction of a second. Are you sure that adding matrix linear constraint is the speed bottleneck?

The new code is still in my dev branch https://github.com/hongkai-dai/drake/tree/add_linear_matrix_constraint

hongkai-dai, 2020-05-14:
@sadraddini Is this still a problem? I think based on the previous comment, AddLinearConstraint is not the speed bottleneck? If not, could you close the issue?

jwnimmer-tri, 2023-05-22:
Closing as "missing information".  If we have new benchmarks are that slow, we can open new issues for them.