RobotLocomotion/drake #12560: Additoinal checks that framework is compatible with Context, State, Parameters, etc.

edrumwri, 2020-01-11:
As came up in #12551, some work still remains in the systems framework to check that:

1. System functions that take in a State should check to see whether the State came from a Context created by that System. See, for example, System::CalcUnrestrictedUpdate(.)

2. Input and output port evaluation methods need to check that the passed in Context was created by a System that declared the port.

3. Cache entry evaluation methods need to check that the passed in Context was created by a System that declared the cache entry.

jwnimmer-tri, 2020-01-11:
Also:

4. SystemConstraint calculations.

edrumwri, 2020-01-13:
6. System::FixInputPortsFrom(.)

jwnimmer-tri, 2020-01-22:
Step 1 will be to move `geometry::Identifier<Tag>` to `//common` and use it instead of a bare `uint64_t`, so that we can decorate even more Context-related types with it without spreading the magic POD disease further.

jwnimmer-tri, 2020-01-22:
WIP for this issue will be at https://github.com/jwnimmer-tri/drake/tree/system_id_more

jwnimmer-tri, 2020-02-13:
Aside (probably for another issue in the future) -- I just made the mistake of passing the diagram context (not plant context) into `MultibodyPlant::SetFreeBodyPoseInWorldFrame`.  Plant methods like that should also add the context-check.

jwnimmer-tri, 2020-11-18:
\CC @rpoyner-tri FYI this is one of the more important usability defects in Drake.  If there's a triage on the way to try to get usability improvements, anyone is welcome to steal this / co-work this with me.

dmcconachie-tri, 2021-02-26:
I believe I just stumbled on this in:

7. Frame::CalcPoseInWorld

jwnimmer-tri, 2021-03-03:
My current checklist to help measure what's done or not:
- [x] All public methods of SystemBase
- [x] All public methods of System, except for:
  - [ ] EvalTimeDerivatives
  - [ ] CheckValidOutput
- [x] All public methods of LeafSystem
- [x] All public methods of Diagram, except for:
  - [ ] DoCalcTimeDerivatives
  - [ ] DoCalcImplicitTimeDerivativesResidual
  - [ ] GetSubsystemDerivatives
  - [ ] GetSubsystemDiscreteValues
  - [ ] GetSubsystemCompositeEventCollection
  - [ ] GetMutableSubsystemCompositeEventCollection
  - [x] GetMutableSubsystemState(Context&)
  - [ ] GetMutableSubsystemState(State&)
  - [ ] GetSubsystemState

rpoyner-tri, 2021-05-04:
Attempting to grok the rather terse checklist above. If I understand things adequately, it looks like the missing SystemOutput checking would involve adding a system_id to the SystemOutput object and implementing a check on it, similar (or identical) to ValidateChildOfContext(). Is that the ask?

jwnimmer-tri, 2021-05-04:
Yup -- ideally, it's literally `ValidateChildOfContext` aka `ValidateCreatedForThisSystem`.  At the same time, we should also stop using (and deprecate) the less-stringent validation method `CheckValidOutput` that is currently used.

In other words, we want this patch to compile and pass existing tests:

```patch
--- a/systems/framework/system.cc
+++ b/systems/framework/system.cc
@@ -439,7 +439,7 @@ void System<T>::CalcOutput(const Context<T>& context,
                            SystemOutput<T>* outputs) const {
   DRAKE_DEMAND(outputs != nullptr);
   ValidateContext(context);
-  DRAKE_ASSERT_VOID(CheckValidOutput(outputs));
+  ValidateCreatedForThisSystem(outputs);
   for (OutputPortIndex i(0); i < num_output_ports(); ++i) {
     // TODO(sherm1) Would be better to use Eval() here but we don't have
     // a generic abstract assignment capability that would allow us to
```

rpoyner-tri, 2021-05-04:
Follow-up question: should CheckValidOutput just go away altogether?

rpoyner-tri, 2021-05-04:
Fun subproblems in the remaining checklist items:

* SystemConstraint<=>System near-dependency cycle
* The {Discrete,Continuous}State objects appear to have a more interesting life cycle than, say, SystemOutput. Will need to trace things through carefully.

jwnimmer-tri, 2021-05-04:
> SystemConstraint<=>System near-dependency cycle

Yeah.  Upon first look, the `SystemConstraint` will need its own error-reporting method, since it can't reuse the `SystemBase` one (nor the `System::ValidateCreatedFor...`, etc.).  However, since it has its own `description` anyway, reporting that on its own seems plausibly good enough.

rpoyner-tri, 2021-05-24:
I updated the checklist above with a few more System methods that are now checkable, thanks to more helper objects having labels.

A few helper types are not yet labeled: `Parameters`, `AbstractValues`, and perhaps some I haven't noticed yet.

rpoyner-tri, 2021-06-07:
Updated the checklist. Looks pretty done to me.

closed 2021-06-07 by rpoyner-tri