RobotLocomotion/drake #12561: Make sure we have adequate testing for integrators in fixed-step mode

antequ, 2020-01-11:
@amcastro-tri commented on PR 12532 that we could have testing for integrators in fixed-step mode. 

Since that PR was mainly meant to refactor the tests for implicit integrators together into one area, I didn't want to fully address the issue of testing fixed-step integrators. 

I am creating this issue as a TODO to look over our tests and see if they already adequately test fixed-step mode integration.

+@sherm1 for visibility, @edrumwri in case he's interested.

antequ, 2020-03-21:
We should also add a setting that enables integrators to run without error estimation (maybe this deserves a separate issue). Currently setting fixed step mode to true simply runs the integrators without error control; they still compute the error estimation. For some integrators this would be easy to add, while for others this might be harder.

jwnimmer-tri, 2024-11-01:
We should either write the test as part of the feature development because it is surely important, or else just wait to write tests until we encounter a real bug. Closing this as "not planned" -- having "missing unit test" bugs open for years is of no value.

closed 2024-11-01 by jwnimmer-tri