RobotLocomotion/drake #12566: Support a namedview workflow in C++.

RussTedrake, 2020-01-13:
In #11191 we introduced this for python.  
Possible related to #9171.

Specific need/request:
Authoring systems is safer/easier if we can work with structured data (as opposed to a BasicVector).  But requiring the port to be a structured type defeats many workflows.  For instance, in my `AcrobotGeometry` PR (coming shortly, will link to this issue), I was fine wiring up the `AcrobotPlant` to the `AcrobotGeometry` system if the input port had type `AcrobotState<double>`, but then I could NOT easily wire up a trajectory playback to visualize the output of trajectory optimization (as done in `run_swing_up_traj_optimization.cc`.

I think a few sugar methods in the auto-generated `*.named_vector` workflow would be able to address this easily.

jwnimmer-tri, 2020-01-13:
#10520 might be a draft to start from.

RussTedrake, 2021-08-01:
Note that the "State variables 2.0" doc in #9171, currently also includes the view concept:
```
// Returns the value of this numerical variable but overlaid with
// a struct to provide named access via members.
template <template<typename> Overlay>
const Overlay<T>& variable.value_overlay<Overlay>(const Context<T>&);
```
Let's keep these two ideas in sync!

closed 2024-11-01 by jwnimmer-tri