RobotLocomotion/drake #12575: No way to "un" fix an input port

edrumwri, 2020-01-14:
Though it's presumably not a common thing to want to do, we have need to do it (we're using a Context as a workspace, we need to fix a value to an input port to be able to evaluate an output port on the corresponding System, and we want to unfix that port so that the value we fixed doesn't inadvertently get used in the future).

The most apparent way to accomplish this would be to pass in a nullptr for the value-to-be-fixed to the appropriate FixInputPort methods in the Context, but that doesn't work (they demand that the value be non-null).

sherm1, 2020-01-14:
Proposed resolution: add an `input_port.UnFixValue()`  method that replaces the previously-fixed value with a nullptr in `ContextBase::input_port_values_`.

sherm1, 2020-01-14:
An alternative would be to allow `input_port.FixValue(nullptr)` but I like the explicit `UnFixValue()` better since it still allows us to catch inadvertent nulls.

sherm1, 2020-01-15:
Upon investigation, this is somewhat trickier than it looks. When an input port is first given a fixed value, a `DependencyTracker` object (with a unique "ticket") is allocated for the stored value, and the input port subscribes to the value's tracker. That way any subsequent changes to the fixed value are propagated properly to downstream dependents of the input port, invalidating now-stale cache entries.

To undo that completely:
- the value should be set to null,
- change notifications should propagate, 
- then the input port should unsubscribe from the value's tracker, and
- the value's tracker should be deleted.

Currently there is no way to re-assign a ticket, so the previously-used ticket will be retired like  Shaquille O'Neal's jersey. So repeated fixing and un-fixing the value of an input port will allocate new tickets repeatedly. This creates an empty slot in the dependency graph (just an std::vector of pointers) so will eat up 8 bytes each time. That seems tolerable but deserves a note in the UnFixValue() docs -- in contrast simply changing the existing fixed value to a new value continues to use the already-allocated tracker and ticket.

jwnimmer-tri, 2020-01-15:
Perhaps worth noting another alternative: clone the original (unfixed) context, and fix the input only in the clone, leaving the original context undisturbed.

edrumwri, 2020-01-15:
@jwnimmer-tri cloning the Context won't work for us- we're maintaining a separate Context as a workspace to avoid heap allocations.

jwnimmer-tri, 2020-01-15:
You can preallocate the clone wherever you are preallocating the original, though?  Or are you saying that SetTimeAndStateAndParametersFrom does new allocations?

edrumwri, 2020-01-15:
Hmm.. This is not one of those places that I anticipated would perform a heap allocation (perhaps it was caching that my mental model of FixInputPort() did not capture?)

edrumwri, 2020-01-15:
> But anyway something else is wrong -- fixing an input port is going to touch the heap. (_"When an input port is first given a fixed value, a DependencyTracker object ... is allocated for the stored value, ..."_.)

Ugh. Heap allocations could seem to be avoided with the dependency tracker by replacing an already-fixed value. But a heap allocation is currently necessary for one of the parameters to ContextBase::SetFixedInputPort().

sherm1, 2020-01-15:
Replacing a fixed value with a new one should be done with `FixedInputPortValue::GetMutableData()` or `::GetMutableVectorData()`, rather than `ContextBase::SetFixedInputPort()`. In that case you can set the value with no heap activity.

jwnimmer-tri, 2020-01-15:
> An alternative would be to allow input_port.FixValue(nullptr) but I like the explicit UnFixValue() better since it still allows us to catch inadvertent nulls.

Another way to spell this would be `FixedInputPortValue::Clear()` or Undo or Reset or etc.  Making it an operation on the object that denotes the fix to the user might be more cohesive.

>    the value should be set to null,
>    change notifications should propagate,
>    then the input port should unsubscribe from the value's tracker, and
>    the value's tracker should be deleted.

Instead of all of this (which leads to the packrat problem), is it possible instead to toggle something with FixedInputPortValue or in the ContextBase, to mark the value as disconnected (as-if it were nullptr)?  At worst, that could be `vector<bool> live_input_port_values_;` in ContextBase.  Then, if the input were re-fixed again in the future, perhaps we could do that without new allocations.

Keeping the objects might also make it easier to give good error messages, e.g., if a user had a FixedInputPortValue reference and tried to call into it after the input was unfixed.

sherm1, 2020-01-15:
> Instead of all of this (which leads to the packrat problem), is it possible instead to toggle something with FixedInputPortValue or in the ContextBase, to mark the value as disconnected (as-if it were nullptr)?

I think so. Now that I understand OP's need to avoid heap allocation, keeping the FixedInputPortValue object around but disabled would be better. This would require a change to the input port evaluation mechanism that we use to determine whether there is a fixed value present, but that shouldn't be any worse than the recipe I outlined above. And in that case we could leave the input port subscribed to the FixedInputPortValue since its DependencyTracker and ticket are still valid, if unused for the time being.

sherm1, 2020-04-16:
@edrumwri is this still a problem for you? I'd like to close this issue because the fix difficulty seems out of proportion to the benefit.

edrumwri, 2020-04-16:
We've worked around it. But are you sure you want to close a known issue? I think just saying that a fix is not in the cards at the moment, like you did above, is fine. The open issue also discusses some workarounds.

sherm1, 2020-04-16:
OK, thanks. Makes sense -- I'll leave it open at a lower priority.

jwnimmer-tri, 2024-11-01:
> But are you sure you want to close a known issue?

For feature requests?  Yes.  If it is not planned, we don't keep the ticket open.

closed 2024-11-01 by jwnimmer-tri