RobotLocomotion/drake #12577: Fix linter to support structured bindings.

amcastro-tri, 2020-01-14:
Among many other useful feature, we can now use C++17's structured bindings in Drake. For instance #12559 updates the return from `AddMultibodyPlantSceneGraph()` so that we can write:
```c++
auto [plant, scene_graph] = AddMultibodyPlantSceneGraph(builder);
plant.DoMultibodyStuff();
scene_graph.DoGeometryStuff();
```
though there might be of course other places where we can use this same feature.

However our linter hates the space between `auto` and the squared bracket `[` and forces us to write instead:
```c++
auto[plant, scene_graph] = AddMultibodyPlantSceneGraph(builder);
```
with no space between `auto` and `[`.

We'd need to update the linter to handle this case. 

Per [this](https://reviewable.io/reviews/robotlocomotion/drake/12559#-LyKyGb60f1t0DCw2y10) review comment a good starting point would be the [work here](https://github.com/google/styleguide/pull/374).

jwnimmer-tri, 2020-01-15:
FYI I have a WIP branch https://github.com/jwnimmer-tri/drake/tree/new-cpplint-upstream to switch us to https://github.com/cpplint/cpplint, which is more actively maintained.  The structured bindings fix was submitted there at https://github.com/cpplint/cpplint/pull/42/commits but was closed without merging.  We could imagine resurrecting that fix and then switching our cpplint to a new upstream.

m-chaturvedi, 2020-02-04:
Looks like they already gotten something similar merged?
https://github.com/google/styleguide/pull/487/

PR for our fork in case we'd like to cherry-pick: https://github.com/RobotLocomotion/styleguide/pull/22

EricCousineau-TRI, 2020-02-04:
I'm down for a simpler fix for now. Will review `styleguide` fork change.

PR #12687 (2020-02-05): cpplint: Update sha and reformat to support structured bindings
Resolves #12577

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12687)
closed 2020-02-05 by EricCousineau-TRI