RobotLocomotion/drake #12592: MultibodyPlant Jacobian methods throw an assertion when Frame A is not the world frame.

edrumwri, 2020-01-16:
This is true at least for CalcBiasForJacobianSpatialVelocity(). I suspect this is also true for CalcJacobianSpatialVelocity(.), but the documentation does not indicate as such.

IIRC, it's fairly straightforward (albeit still not frictionless) to change Frame A to the world frame, but how to do so should be documented, at minimum. Ideal, of course, would be to actually make real use of the Frame A argument!

I'm marking this as high priority, since these otherwise beautiful methods are halfway broken (halfway working for the optimist), and because I actually want to use this feature.

sherm1, 2020-01-16:
Queued up for implementation asap.

mitiguy, 2020-03-03:
@edrumwri   I just started PR #12804 that fixes a related Jacobian problem and then will proceed with this issue #12592.  The good news is that it seems as if CalcJacobianSpatialVelocity() is working now with whatever frame A you want to use.. However, as you found, CalcBiasForJacobianSpatialVelocity() is  not working for an arbitrary frame A.

edrumwri, 2020-03-03:
Got it- thanks for the status update Paul!

mitiguy, 2020-04-27:
@edrumwri  PR #13063 renames CalcBiasForJacobianSpatialVelocity() to CalcBiasSpatialAcceleration() and renames CalcBiasForJacobianTranslationalVelocity()  to CalcBiasTranslationalAcceleration().  
It implements new notation per issue #12140 with corresponding documentation.
It includes some other small changes that are helpful in this PR -- and will make it easier to  address upcoming issue #12592, submitted by user Evan Drumwright -- CalcBias methods throw assertions if measured-in-frame Frame A is not the world frame.

mitiguy, 2020-06-22:
@edrumwri  A meaningful part of issue #12592 has been addressed since PR #13453 merged and CalcBiasSpatialAcceleration() should now work with whatever "measured-in" frame you like.  I am now working on CalcBiasTranslationalAcceleration() so that should work similarly.  I am also sending this via email to edrumwri@gmail.com.

closed 2020-07-08 by mitiguy