RobotLocomotion/drake #12594: Port OpenGL renderer to Drake

tehbelinda, 2020-01-17:
Was added to Anzu due to build issues in Drake (@SeanCurtis-TRI if you could elaborate?).  

We'd like to bring it back into Drake and add more features so it's comparable to RenderEngineVtk, namely adding color and label images. The main benefit is that we believe the purely OpenGL based renderer will be a faster implementation compared to those with VTK. We will also be able to better profile the OpenGL calls and benchmark it against the other render engines to confirm this theory

sherm1, 2020-01-17:
Assigning to Sean first for elaboration.

sherm1, 2020-01-17:
The goal here is to have a faster rendering option so performance measurement & optimization are an important part of the project. This may be an Ubuntu-only feature due to problems with Mac OpenGL.

SeanCurtis-TRI, 2020-01-17:
Yep. What @sherm1 said. We'll have to put compilation guards in place as Mac no longer supports OpenGL.

jwnimmer-tri, 2020-01-22:
@tehbelinda asked me on slack about prior examples of Ubuntu-only features.  Here's an elaborated copy of my reply:

---

No, we do not have any examples of that.  So far, we have purposely chosen not to fracture Drake support by platform, because it gets expensive fast.  (Maybe in this case it's worth it.)

At https://github.com/RobotLocomotion/drake/blob/master/systems/sensors/godot_renderer/dev/BUILD.bazel there is a little bit of syntax that might help get us started, to omit files from the build step.

(Aside -- we should probably purge godot_renderer folder from Drake anyway; there are no active plans to move it back from Anzu in the near term to my knowledge.)

Another example is https://github.com/RobotLocomotion/drake/blob/master/solvers/BUILD.bazel where we have snopt_solver.h, snopt_common.cc, snopt_solver.cc, no_snopt.cc where the same header file has two different implementations -- one that works, and one a stub that always throws an exception.

However, our job for gl_renderer is going to be a nightmare no matter what.  Telling bazel to skip compiling the files on apple is only moderately difficult -- the tough part is going to be all the places that want to _use_ the code are going to have to have platform-specific conditionals somehow, e.g., any tests that depend on it are going to have to be altered or excluded on apple.  Tests within the gl_renderer folder will only be a few and not so hard since they are 100% skipped.  However, then any examples or application benchmarks are going to be infected with a test guard, etc. throughout the tree, even if their dependency on the GL renderer is only an aside.

We will also either have to guard call site include statements with the preprocessor, or we'll have to do the stub-file thing as we do with snopt, which means updating gl_render_engine.h to use Pimpl or something so that it does not have any in-package includes that we'd also have to stub out.  Or maybe a factory-method wrapper instead.

Since the only reason to add a faster renderer would also be to install it for users, we may also need to tweak any install rules or build_components logic to support installing Ubuntu-only code.  I can't quite predict whether that will work out of the box or not.

Since the Ubuntu-conditionalizing is the first _new_ thing about this code beyond what's in Anzu, I suggest doing that first in its own PR.  We could add `class RenderEngineGl final : public geometry::render::RenderEngine {` class in Drake, but whose `RenderDepthImage` returns some bogus constant data on Ubuntu (+ unit test!), and where the build system is set to only support the feature on Ubuntu.  That would let us prove out an de-resk the conditionalizing, ahead of the other work on this issue.

jwnimmer-tri, 2020-05-01:
@SeanCurtis-TRI do we have a "catcher" or "beta tester" identified for this feature?

SeanCurtis-TRI, 2020-05-01:
A fair question, one for which I have no real answer. 

The logical catcher would be people using the source version today. However, it's hard to justify their effort to swap from one incarnation to the other when nothing of value has changed for them. When label and color images are added, then they become obvious catchers, as concrete value will justify the effort to change which incarnation is being exercised.

Thoughts?

jwnimmer-tri, 2020-05-05:
> The logical catcher would be people using the source version today.

By this, do you mean the TRI Manipulation team (Anzu)?  I concur we would be excited to try new features, but those additions could have been developed on the internal copy instead, with an even tighter feedback cycle and many fewer CI headaches.

Do we have any non-Anzu catchers lined up?  In particular for graphics-related features, the installation / distribution / packaging pipeline challenges are not really exercised by Anzu use, since we do it ourselves.

I think it would help if you had an external user in mind that would be willing to help acceptance-this code as it crosses over into master, so that we can be sure its actually successful.  At worst, a non-Anzu TRI internal user.

SeanCurtis-TRI, 2020-05-05:
Yes. Other internal TRI users. Everyone needs faster rendering.

SeanCurtis-TRI, 2020-06-30:
I'm going to open a separate issue for missing features, leaving *this* PR to be strictly about porting from Anzu (which will be accomplished by #13616).

PR #13616 (2020-07-01): [render] Move RenderEngineGl out of dev into master
This is a transfer of feature-reviewed code in dev/ into mainstream. This the last of the RenderEngineGl code as part of the port. With this PR `gl_renderer/dev` goes away.

resolves #12594

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13616)
closed 2020-07-01 by SeanCurtis-TRI