RobotLocomotion/drake #12598: Deprecate legacy parsing of contat properties in URDF and SDF

SeanCurtis-TRI, 2020-01-17:
PR #12491 changed how friction coefficients are to be defined for drake applications. Currently, when parsing SDF files, we hijack the `<surface><friction><ode>` values. In URDF files we use the "drake_compliance" tag to get friction, youngs_modulus and dissipation.

Those have all been supplanted by the new, uniform paradigm. We need to deprecate and then remove this dependency. Because it's in parsing, it's not a straight-forward compile-time deprecation. It'll have to be one-time warnings when parsing files.

RussTedrake, 2021-12-05:
I'd like to understand this a little better.  On the surface, I strongly disagree.

My perspective: I find myself very often explaining to people why the models that they find in another simulation engine/environment don't work correctly in Drake out of the box.  I believe Sherm's principle of least surprise would dictate that we can add custom tags for beautiful specification of contact parameters if we like, but we should ALSO support the default tags that other engines are using.  This seems to be taking a step in the wrong direction.

If we agree, then the TODO in the code should be removed: https://github.com/RobotLocomotion/drake/blob/cfd90f9e9e2ed0e29e07b97fb152be0f0c61ec8e/multibody/parsing/detail_scene_graph.cc#L432-L433

SeanCurtis-TRI, 2021-12-06:
So, my arguments are as follows:

1. The behavior this issue addresses is hijacking *some* (but not remotely all), of settings that are explicitly targeted toward a particular software package: `ode`. By parsing *this* we imply support for `ode` but, we've never promised it and it's woefully incomplete.
    - For the record, we use `<mu>` and `<mu2>` to mean coefficients of static friction and dynamic, friction, respectively. This is *not* [what sdformat documents them to be](http://sdformat.org/spec?ver=1.8&elem=collision). `mu` is: "Coefficient of friction in first friction pyramid direction, the unitless maximum ratio of force in first friction pyramid direction to normal cone". (And mu2 is similarly defined for the *second* direction.) So, while we're using previously existing *tags*, we're not interpreting them to mean the same thing.
2. SDF supports parallel specifications. Instead of `ode` we might have randomly picked `bullet` instead. `<surface><friction><bullet><friction>` could've served as well as the `<ode>` value. But Drake is neither bullet, nor ode. Should a user have the expectation that bullet successfully consumes ode parameters or the other way around?
3. By using *only* our custom traits, we unify specification across URDF and SDF. If we continue with the current hijacking behavior, we end up with radically different mechanisms between the two formats. This probably isn't a *huge* problem because we'd assume that any single user would only use one format and not be sensitive to the cross-format compatibility.
4. If we support *some* `ode` flags? How do we pick which ones we support and which ones we don't? Should we support all of them?
5. If we support `ode` flags, should we also support `bullet` flags? And if we support multiple redundant flags (as we currently do), what type of protocol should we use to reconcile conflicting values?

SDF has, built in, simulator-specific parameters. It opened this Pandora's box. If the box were closed, it would make sense to simply use canonical SDF syntax. But with the box wide open, there's no such thing as canonical SDF syntax. One logical thing to do is carve out a corner of the world for ourselves.

So, we have many options (here are the first few that rolled off the top of my head):
  - Simplify the parsing code and only support one specification of physical properties
      - We can  with the using the `<ode>` specification + `<drake:...>` specification.
          - In this case, we should *carefully* document what is supported and what *is not*.
      - We can just use `<drake:...>` specification.
          - This likewise requires explicit documentation.
  - We continue with the status quo where we support multiple modes of specification.
      - Carefully document when one overrides the other.
  - We come up with a software solution where the user can say, "use my ode values" or "use my bullet" values.
      - In my mind, this would be related to, but distinct from, @jwnimmer-tri's proposals on handling parsing errors/inconsistencies (where an error-handling protocol is passed into the parser by the user).
      - But in any case where those tool parameters are insufficient for Drake, they'll still have to augment with Drake-specific quantities.

Ultimately, this is a complex thing. And this is strictly about *friction coefficients* -- it only gets worse when we look at other parameter values. The only thing that is clear to me is that simply keeping the status quo is not obviously a good thing. It requires defending and shoring up if it is deemed to be good. My particular value systems suggests it's not a good thing.

sherm1, 2021-12-06:
Great points -- thanks!

jwnimmer-tri, 2021-12-06:
My general ideas here would push us as follows:

When Drake receives an input file (SDFormat, in this case), then if there are parts of the file that we ignore (like the ode or bullet information), then _at minimum_ we must always warn the user that we are ignoring them.  Any extra junk in an input file _must never_ be silently ignored [1].  That would mitigate, for example, the `<bullet/>` element being ignored.

As far as the "ODE mu" and such go -- we should think of that like "importing" a foreign format, something along the lines of importing a MuJoCo file.  We could offer an offline conversion tool to convert an SDFormat file with bullet physics into a Drake-native format (with Drake elements).  Or, if the support for bullet-to-drake conversion is sufficiently robust, we could even have the conversion happen automatically.  What we should _not_ do is automatically load bullet-specific (or ode-specific) files incompletely, and yet pretend to the user that the loading of it was complete and correct.  So the conversion can only be implicit to the extent it is sufficiently comprehensive and will lead to a similar physical outcome.

---

[1] (We should possibly also turn the junk turn into hard errors, instead of just warnings, maybe under a parser runtime flag.  But at minimum, we need some kind of warning, to start.)

RussTedrake, 2021-12-06:
I think we should definitely warn / throw on some tags which we know about, and for which we know that ignoring them can lead to errors.  I'm less sure that I agree about warning/throwing on every unknown tag.  That seems against the guidelines of XML.  (I was trying to find a reference, but the "XML Elements are Extensible" section of [this](https://www.w3schools.com/xml/xml_elements.asp) is at least an indicator).

jwnimmer-tri, 2021-12-06:
Things in third-party namespaces should be silently ignored, yes.

Things declared the in sdformat namespace but that are not part of the sdformat schema should be errors.

Maybe for sdformat things like "actor" or "plugin" that are defined by the spec, but orthogonal to Drake, we can make the case for silently ignoring them, or at least have them be a different level of warning/error than the mechanically-related format elements.