RobotLocomotion/drake #12604: Inconsistent class / constructor documentation in pydrake

RussTedrake, 2020-01-19:
Related to #11954.

I've noticed that the documentation for the class and the constructor of ManipulationStation is missing:
https://drake.mit.edu/pydrake/pydrake.examples.manipulation_station.html?highlight=manipulationstation#pydrake.examples.manipulation_station.ManipulationStation

By contrast, the Adder system looks good:
https://drake.mit.edu/pydrake/pydrake.systems.primitives.html?highlight=adder#pydrake.systems.primitives.Adder_

Is this inconsistency still expected, given that #11954 is closed?

EricCousineau-TRI, 2020-01-23:
It doesn't seem expected; the docstring seems like it should be there:
https://github.com/RobotLocomotion/drake/blob/489048bd4fae938df4cbb850d6dff04522b86101/bindings/pydrake/examples/manipulation_station_py.cc#L40-L41

\cc @jamiesnape

m-chaturvedi, 2020-02-03:
-  > I've noticed that the documentation for the class and the constructor of ManipulationStation is missing:

   We might have missed it because it's not at its usual place:
   http://bit.ly/2Un1rMI
   [ Converted to bitly because it looks like there's a markdown bug for hyperlinks with trailing underscores : https://github.com/markdown-it/markdown-it/issues/38 ]

-  Why is this happening?

    It looks like in Python's `sorted` function which is used to sort the doc members alphabetically,  capital letters are favored over underscores:
    https://repl.it/repls/BusyEmbellishedPostscript 
    https://github.com/sphinx-doc/sphinx/blob/master/sphinx/ext/autodoc/__init__.py#L501-L506
    (Ubuntu's Sphinx 1.6.7 has a similar `__init__` file that I'd tested with)   

-  How to 'fix' it?

   One way that I've tried, could be:  https://github.com/m-chaturvedi/sphinx/commit/45476ed5cfd056b26deac25d7c03430db72ca1ab

RussTedrake, 2020-02-04:
Thank you. What about the class documentation?

https://drake.mit.edu/doxygen_cxx/classdrake_1_1examples_1_1manipulation__station_1_1_manipulation_station.html

m-chaturvedi, 2020-02-04:
It looks like we missed the docstrings for them (#12675).

EricCousineau-TRI, 2020-02-04:
Regarding member order: Hm... I'm really hesitant about forking `sphinx` code; would almost prefer monkey patching over that.

That being said, it looks like all classes have their `__init__` members placed in the same order.

Looks like there are options to control the ordering, e.g.:
https://stackoverflow.com/a/58996596/7829525
Perhaps there's a way to intercept that with a handler / monkey patch instead.

m-chaturvedi, 2020-02-05:
> Regarding member order: Hm... I'm really hesitant about forking sphinx code; would almost prefer monkey patching over that.

I was meaning to create an issue and seeing if they accept the issue / PR.  Maybe that wouldn't solve our problem because we're using an Ubuntu package.  

I had tried the `bysource` option in `config.py`, but it had not solved our problem.

Another way could be to more consistently name our methods.  Please correct me if I'm wrong, but I guess our (regular) python code style doesn't accommodate method/functions starting with a capital letter.  Maybe we should not care about the C++ style in our bindings' code and make it consistent with regular python code by making those function names snake_case.

jamiesnape, 2020-02-06:
> Another way could be to more consistently name our methods. Please correct me if I'm wrong, but I guess our (regular) python code style doesn't accommodate method/functions starting with a capital [...]

I would probably go the opposite way and change pure Python public methods into non-snake-case (any other Python in Drake follows normal naming conventions, however), but honestly, we should not rename anything while we are working out generating the bindings.

jwnimmer-tri, 2020-02-06:
I don't think anyone is complaining about the order of CamelCase vs snake_case methods.  The only complaint is that `__init__` should be at the top, yes?

EricCousineau-TRI, 2020-02-06:
That's my complaint (and I believe Russ's). I do not believe we should change any of the conventions used in the Python bindings.

EricCousineau-TRI, 2020-02-06:
Per f2f, this might be happening because of the issue in https://github.com/RobotLocomotion/drake/issues/11954: B/c `pybind` does not produce a full name that includes the class name, we then introduced a workaround to always put `__init__` in, but then we may affect the sorting order because of this.

Ideally, we just find a way to plugin to change the sorting order.

RussTedrake, 2020-02-09:
Yes, `__init__` at the top was my primary ask (beyond the missing class documentation which was already resolved).

m-chaturvedi, 2020-02-11:
I'd asked it on the group: https://groups.google.com/forum/m/#!msg/sphinx-users/bSAWjiiDWik/BlRNSMQoFgAJ

EricCousineau-TRI, 2020-02-11:
In the Group post, I think it'd be important to mention the lines you alluded to - where we provide an explicit exception for `__init__` capsules:
https://github.com/RobotLocomotion/drake/pull/12452/files

If we hadn't needed that exception, I'd assume everything would be fine.

m-chaturvedi, 2020-02-12:
Sorry for the late response. 

> Per f2f, this might be happening because of the issue in #11954:

No, I am not sure about this.

>In the Group post, I think it'd be important to mention the lines you alluded to.

I am not sure about this either.

I think the two problems are disparate, at least at the outset.  I can demonstrate my claim with the following (tested using `tox -e py36 tests/test_autodoc.py`):
https://github.com/sphinx-doc/sphinx/compare/master...m-chaturvedi:test_member_order?expand=1

I maintain that the cause is: `assert sorted(["a", "__init__", "A"]) == ["A", "__init__", "a"]`.

EricCousineau-TRI, 2020-02-12:
Yup, that looks like pretty solid proof that it's just sorting. Thanks for posting it!

EricCousineau-TRI, 2020-02-13:
As Mmanu showed, `bysource` does the correct thing for Python code, but it's unclear if this will work for Python C extensions.

m-chaturvedi, 2020-02-17:
IIRC, the code analyzer which doesn't seem to work for us:
https://github.com/sphinx-doc/sphinx/blob/af62fa61e6cbd88d0798963211e73e5ba0d55e6d/sphinx/ext/autodoc/__init__.py#L674

`self.analyzer` is `None` when `bysource` is used for our code.

There have been other complaints regarding `bysource` not working properly:
https://github.com/sphinx-doc/sphinx/issues/3673

EricCousineau-TRI, 2020-03-19:
Re-opening; see #12910 for details.

PR #12921 (2020-03-23): pybind11 docs: Fix sphinx patch for versions > 1.6.7
Towards: https://github.com/RobotLocomotion/drake/issues/12604

PR #12787 was reverted (#12910), due to failures on Mac.
This PR fixes them.

The Mac failures can be reproduced on Linux by:
- `apt-get remove python3-sphinx`.
- `python3.6 -m pip  install sphinx sphinx_rtd_theme`
- `bazel test --test_output=all --nocache_test_results //bindings/pydrake/doc:gen_sphinx_test`

The first commit `652aa39`, has already been reviewed in #12787 .
Please review the  the second one.

Closes #12604 

---
This change is [[image]](https://reviewable.io/rev [... cut]
closed 2020-03-23 by EricCousineau-TRI