RobotLocomotion/drake #12611: [parsing] Versioning "drake:"-namespaced tags and attributes

SeanCurtis-TRI, 2020-01-21:
(Much of this came from [this slack conversation](https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1579617517002200).)

# The problem

We have a fair amount of data (and it [promises to increase](https://github.com/RobotLocomotion/drake/issues/12593)) of Drake-specific data which we parse from our SDF/URDF input files. We've recently adopted the standard of making sure it is properly prefixed with a `drake:` namespace (examples: the tags `<drake:declare_convex/>` and `<drake:capsule>`). We need to have a plan and mechanism in place that will allow for the natural evolution of
  - what exists in the namespace
  - what the contents of the namespace mean

# The propsoal

TBD but with thoughts in the slack conversation.

SeanCurtis-TRI, 2020-01-21:
cc @jwnimmer-tri @ggould-tri

ggould-tri, 2020-01-21:
Scribing some hard-earned lessons from my time at Endeca:
 * Proper xmlns-declared namespaces are good, and necessary for using validating XML parsers.  However
 * XML schema/dtd versioning is an uncomfortable fit for evolving application semantics.  Because XML namespaces cannot reference, descend from, or update one another, it is very difficult to evolve structural elements of the schema by eg adding or removing tags or attributes without creating a proliferation of namespaces.
 * As such semantic versioning should be solved out-of-band of the xmlns mechanism.  There are a million ways to do this; the three we explored at Endeca were:
   * Backward compatibility -- All documents implicitly reference the current HEAD schema.  Solemnly swear that all future schemas will accept all input valid under all past schemas.  You can write an xsd or dtd if you like, but it gets fiddly.  This is easy but breaks down with changes to the meaning of elements (especially with changes to defaults, which give silent, spooky failures!).
   * Forward compatibility -- Automated conversion is available (and tested) from all past versions to the current version.  The applicable semantics version for a document can be automatically detected or specified in an element.  Nobody ever does schema validation on the namespace (ie, you don't write an xsd or dtd).  This is harder but allows nearly any change apart from total abandonment of a feature.
   * Full semantic versioning -- Documents carry a major/minor/release semver element for each xmlns they mention.  Relating them to published and archived protocol documentation and interpreted according to semver standards.  Heavyweight and really only needed for frequent releases and major changes.  Makes your xsd a mess if you ever publish an xsd for your namespace -- you'll probably need to write xsd-generating tools if you decide to care.

IMO we're likely to end up at the forward compatibility style of solution.

tehbelinda, 2020-01-22:
I'm also inclined towards forward compatibility as it's still easy for the user as well as cleaner for developers, e.g. being able to compartmentalise the conversion code

EricCousineau-TRI, 2020-01-27:
FTR Sean, @azeey, @scpeters, and myself also discussed this, but only for SDFormat (and the `libsdformat` parser).

rpoyner-tri, 2026-11-13:
Closing as not planned. This was filed in 2020 and the discussion never converged on a versioning scheme for `drake:` tags. Nothing has changed in the years since. If a concrete compatibility problem comes up, please open a new issue describing it.

closed 2026-11-13 by rpoyner-tri