RobotLocomotion/drake #12613: Actuators are not added to joints loaded from URDF, works fine with SDF.

mattcorsaro1, 2020-01-21:
## Problem
My URDF (see #12612 and my [robotiq_3f_WIP branch](https://github.com/mattcorsaro1/drake/tree/robotiq_3f_WIP)) looks fine in the geometry inspector, but actuators are not added to the joints when loaded `multibody::Parser(&plant).AddModelFromFile(full_name)` is called to load the URDF. When it's called to load an SDF generated from that URDF, the model is loaded correctly.

## Steps to reproduce
With my branch, run:
```
bazel-bin/examples/robotiq_3f/run_robotiq_3f_constant_load_demo_URDF
```
The error I get is:
```
abort: Failure at examples/robotiq_3f/run_robotiq_3f_constant_load_demo_URDF.cc:92 in DoMain(): condition 'plant.num_actuators() == 11' failed.
Aborted (core dumped)
```
`plant.num_actuators()` returns 0.

I then used Gazebo to convert my URDF to an SDF:
```
gz sdf -p robotiq-3f-gripper_articulated.urdf > robotiq-3f-gripper_articulated.sdf
```
`examples/robotiq_3f/run_robotiq_3f_constant_load_demo.cc` is a copy of `examples/robotiq_3f/run_robotiq_3f_constant_load_demo_URDF.cc` that loads this SDF instead.
It loads the model as expected:
```
bazel-bin/examples/robotiq_3f/run_robotiq_3f_constant_load_demo
```

sammy-tri, 2020-01-22:
In drake, joints which are specified without a corresponding transmission element are modeled as unactuated joints (It's not clear from just https://wiki.ros.org/urdf/XML is this is the "correct" behavior, or even if "correct" is a meaningful term here), however when drake parses an SDF, any joints with a non-zero force limit are modeled as actuated joints.

In this urdf, we have joints with effort limits but no transmissions, so no actuators are created.

Any opinions on the correct behavior for this case from the community?

In the meantime, try adding a block like this to the urdf and let me know if things start behaving.
```
  <transmission name="finger_1_transmission_1">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="finger_1_joint_1"/>
    <actuator name="finger_1_actuator_1"/>
  </transmission>
  <transmission name="finger_1_transmission_2">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="finger_1_joint_2"/>
    <actuator name="finger_1_actuator_2"/>
  </transmission>
  <transmission name="finger_1_transmission_3">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="finger_1_joint_3"/>
    <actuator name="finger_1_actuator_3"/>
  </transmission>
  <transmission name="finger_2_transmission_1">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="finger_2_joint_1"/>
    <actuator name="finger_2_actuator_1"/>
  </transmission>
  <transmission name="finger_2_transmission_2">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="finger_2_joint_2"/>
    <actuator name="finger_2_actuator_2"/>
  </transmission>
  <transmission name="finger_2_transmission_3">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="finger_2_joint_3"/>
    <actuator name="finger_2_actuator_3"/>
  </transmission>
  <transmission name="finger_middle_transmission_1">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="finger_middle_joint_1"/>
    <actuator name="finger_middle_actuator_1"/>
  </transmission>
  <transmission name="finger_middle_transmission_2">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="finger_middle_joint_2"/>
    <actuator name="finger_middle_actuator_2"/>
  </transmission>
  <transmission name="finger_middle_transmission_3">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="finger_middle_joint_3"/>
    <actuator name="finger_middle_actuator_3"/>
  </transmission>
  <transmission name="palm_finger_1_transmission">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="palm_finger_1_joint"/>
    <actuator name="palm_finger_1_actuator"/>
  </transmission>
  <transmission name="palm_finger_2_transmission">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="palm_finger_2_joint"/>
    <actuator name="palm_finger_2_actuator"/>
  </transmission>
  <transmission name="palm_finger_middle_transmission">
    <type>transmission_interface/SimpleTransmission</type>
    <joint name="palm_finger_middle_joint"/>
    <actuator name="palm_finger_middle_actuator"/>
  </transmission>
```

mattcorsaro1, 2020-01-22:
Makes sense. I added transmissions to my URDF and it's loaded properly. Thanks, Sammy.

sammy-tri, 2020-01-22:
Closing this issue as the immediate issue seems to be resolved (yay!).  If we decide to change the drake parsing behavior, we can open a new issue for that.

closed 2020-01-22 by sammy-tri