RobotLocomotion/drake #12616: Diagrams don't report state counts correctly

sherm1, 2020-01-21:
When a Diagram is asked size information about its state it should sum up the size information of its constituent Systems recursively. Diagrams do this correctly for `num_continuous_states()`, but fail for the other counts:  
```
num_discrete_state_groups()
num_abstract_states() 
num_numeric_parameter_groups() 
num_abstract_parameters()
```
all of which are implemented in SystemBase and thus report only locally-allocated variables for the Diagram (of which there are none currently).

Also, `num_constraints()` is implemented in System and also looks suspicious.

cc @RussTedrake

jwnimmer-tri, 2020-01-21:
> ... num_constraints() is implemented in System and also looks suspicious.

`Diagram::Initialize` copies constraints from the subsystems into the `Diagram` itself, so I think it should be fine.

RussTedrake, 2020-01-22:
N.B., this was initially caught with the following test which fails if placed in diagram_test.cc (around line 1951).  

```
GTEST_TEST(DiscreteStateDiagramTest, NumDiscreteStateGroups) {
  DiagramBuilder<double> builder;
  builder.template AddSystem<SystemWithDiscreteState>(1, 2.);
  const auto diagram = builder.Build();
  const auto context = diagram->CreateDefaultContext();
  EXPECT_EQ(diagram->num_discrete_state_groups(),
      context->num_discrete_state_groups());
}
```

RussTedrake, 2020-01-22:
Please resolve the TODO(#12616) in system.h when this is resolved.

PR #12701 (2020-02-25): Fixes declared state and parameter counts for Diagram
This PR introduces a SystemBase data member to record the Context partition sizes needed to hold all declared state and parameters in a Diagram. A protected pure virtual method is used at Diagram construction to total up all the Context sizes of its subsystems. The SystemBase methods
```c++
num_discrete_state_groups()
num_abstract_states() 
num_numeric_parameter_groups() 
num_abstract_parameters()
```
are reimplemented using the new data member.
```c++
num_continuous_states()
```
moves up to SystemBase and is reimplemented.

Per #12616, thi [... cut]
closed 2020-02-25 by sherm1