RobotLocomotion/drake #12620: Overriding `DoCalcNextUpdateTime()` might have no effect.

amcastro-tri, 2020-01-22:
From f2f with @sherm1, if we override `DoCalcNextUpdateTime()` to return a non-infinity value of `next_update_time`, our simulator should guarantee that time goes exactly through `next_update_time` whether new events were scheduled or not by `DoCalcNextUpdateTime()`. It seems that most of our overrides in Drake do schedule an event (see `LcmLogPlaybackSystem`, 
`LcmSubscriberSystem`).

However `Simulator` uses this time value only if something was scheduled, see [here](https://github.com/RobotLocomotion/drake/blob/cdc37f451e2c97e32fb6670f96ebe0daae70a67a/systems/analysis/simulator.h#L1154). That value of time, in `next_timed_event_time_`, gets also passed to the integrators [here](https://github.com/RobotLocomotion/drake/blob/cdc37f451e2c97e32fb6670f96ebe0daae70a67a/systems/analysis/simulator.h#L1163), however [it gets never used](https://github.com/RobotLocomotion/drake/blob/cdc37f451e2c97e32fb6670f96ebe0daae70a67a/systems/analysis/simulator.h#L1452).

As a case where no new events are scheduled, we have `LcmInterfaceSystem::DoCalcNextUpdateTime()`, which might indicate a bug.

sherm1, 2021-02-16:
This came up in real life recently, see #14644. Bumping priority to high.

sherm1, 2021-02-16:
FYI @edrumwri for advice

sherm1, 2021-02-16:
I'm working on a fix. Making a proper unit test is likely to be the hard part.

PR #14663 (2021-02-24): Make sure DoCalcNextUpdateTime() timed trigger is accompanied by an Event object.
When an overridden `DoCalcNextUpdateTime()` reported a trigger time but didn't return any Event object to serve as a handler, the trigger time was being ignored with time advancing as though nothing happened. But that could cause time-triggered events that should have occurred later in that same step to be missed entirely. LcmInterfaceSystem wasn't returning an Event object from its DoCalcNextUpdateTime() override and caused problems.

This PR changes the behavior to treat a no-Event timed trigger as an error. A  [... cut]
closed 2021-02-24 by sherm1