RobotLocomotion/drake #12631: FirstOrderLowPassFilter system slows down a discrete system significantly

huihuaTRI, 2020-01-24:
I have a discrete system with about 40 DoF that can run at 0.9 real-time-factor. The time step is set to 2ms. After adding the FirstOrderLowPassFilter to the input port of the system, the simulation slows down significantly to 0.5 real-time-factor. The size of the desired state is 66.

Then I switched to use the Runge-Kutta-2 integrator that has a similar update behavior as the discrete systems. I set the maximum time step of the integrator to be the same as the discrete update time step. The simulation runs at 0.65 real-time-factor.

To be more clear, here are the diagrams before and after adding the FirstOrderLowPassFilter.

Before adding the FirstOrderLowPassFilter,
> desired state
------------------>  Discrete System

After adding the FirstOrderLowPassFilter
> desired state
------------------>  FirstOrderLowPassFilter   ---------->  Discrete System

Some initial discussion can be found in this slack [thread](https://tri-internal.slack.com/archives/C2BMF3K28/p1579815662008900).

sherm1, 2020-01-25:
Set to low priority for now since Huihua has a workaround for now -- he hacked together a discrete filter internal to his system. But the behavior is surprising and may indicate a bug somewhere.

amcastro-tri, 2020-01-27:
@huihuaTRI, it'd be nice having a bare minimum case reproducing this problem that you could easily share with us. I wonder if adding a low pass filter to examples/simple_gripper or any of the other examples in Drake would reproduce this.

huihuaTRI, 2020-01-30:
Here is the [branch](https://github.com/huihuaTRI/drake/tree/013020_filter_debug)  in which I added the low pass filter to the kuka system to filter the position. Before adding the filter, I could run at maximum of 18.5 real-time-factor. After adding the filter, the maximum real-time-factor is 13.2.

huihuaTRI, 2020-06-12:
What would happen if I mix a discrete system with a continuous system?  For this issue, the FirstOrderLowPassFilter has defined continuous states, which I assume renders this system for continuous system only. Is this true?

sherm1, 2020-06-12:
Drake assumes all systems are "hybrid systems", meaning they mix continuous and discrete state. If you want to understand what that means precisely, read [How the simulation is stepped](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_simulator.html).

jwnimmer-tri, 2024-03-16:
I'm going to guess that #21134 was the root cause here, so I'll call this "probably fixed, fingers crossed".

closed 2024-03-16 by jwnimmer-tri