RobotLocomotion/drake #12632: python using RandomGenerator

RussTedrake, 2020-01-26:
Although we have bound `RandomGenerator` in python, we've never worked out the right way to use it if we are trying to generate random numbers in python.  But it's needed.I think the right solution is pretty natural, but the mechanics are a bit hairy.  

Right now, I can do
```
from pydrake.common import RandomGenerator
from numpy.random import RandomState

generator = RandomGenerator()
rs = RandomState(generator())

print(rs.uniform())  # etc.
```
But I think this is not sufficient, because it is creating a new `numpy.random.BitGenerator`, using the integer output of our c++ generator as the seed.  There is nothing to prevent the python random generator and the c++ generator to proceed in lock-step with each other, introducing unintended correlations.

I think the *right* solution is to do
```
from pydrake.common import RandomGenerator
from numpy.random import RandomState

generator = RandomGenerator()
rs = RandomState(generator)   #   <--- parens removed here 

print(rs.uniform())  # etc.
```
if we can make `drake.common.RandomGenerator` implement the `numpy.random.BitGenerator` interface.  This would allow `numpy.random.RandomState` (and eventually `numpy.random.Generator`, when we move to numpy 1.17) to actually use the c++ generator as it should.

I found a little about it [here](https://docs.scipy.org/doc/numpy/reference/random/extending.html#new-basic-rngs) and some of the referenced  [.pyx code here](https://github.com/numpy/numpy/blob/master/numpy/random/_mt19937.pyx#L128)

There is one more detail: in c++ we always use the default seed of RandomGenerator to produce deterministic outputs.  In python, the default behavior is that the random seed, if `None`, is [pulled from `/dev/urandom`](https://docs.scipy.org/doc/numpy/reference/random/legacy.html#numpy.random.mtrand.RandomState).  We should document this clearly.  And if users do want random runs, then give them a recommended workflow.  I'm using 
```
generator = RandomGenerator(int(time.time()))
```
for today, but the pseudo-random number people will laugh at me if they see that.  I'm sure there is a better way.

jwnimmer-tri, 2020-01-27:
Could you illustrate an example use case where the exact reuse / state-sharing effects come into play?  I am able to follow the approaches laid out above but not yet clear on the motivations.

In particular:
> But I think this is not sufficient, because it is creating a new numpy.random.BitGenerator, using the integer output of our c++ generator as the seed. There is nothing to prevent the python random [...]

A seed is different than the mt19937 internal state.

See [numpy's mt19937.c](https://github.com/numpy/numpy/blob/master/numpy/random/src/mt19937/mt19937.c) how `RandomState(seed=foo_int)` converts the seed is to state in numpy, or see [C++ overload (2)](https://en.cppreference.com/w/cpp/numeric/random/mersenne_twister_engine/mersenne_twister_engine) for how `std::mt19937` initializes state from a seed.

I think it's not be unusual to start an experiment from a single `seed` set on a single master PRNG, and then derive separately-running child PRNGs from the master PRNG by turning master outputs into new child seeds.  In fact, that's how to isolate the ripple effects when the number of random reads[1] vary as an implementation evolves.  For example:

```
master_prng = mt19337(command_line_seed)
plant_prng = mt19337(master_prng.read())
sensor_prng = mt19337(master_prng.read())
plant = CreatePlant(plant_prng)
sensor = CreatePlant(sensor_prng)
```

Here, for a given `command_line_seed` the sensor noise details will remain the same even across code changes to `CreatePlant` that change how many reads are taken from `plant_prng`.

> ... unintended correlations ...

Consider these two examples:
```
A_prng = std::mt19337(fixed_seed)
B_prng = numpy.random.RandomState(seed=A_prng.read()) 
x = [A_prng.read(), A_prng.read(), A_prng.read(),
     B_prng.read(), B_prng.read(), B_prng.read()]
```
and
```
A_prng = std::mt19337(fixed_seed)
A_prng.read() // discard one
y = [A_prng.read(), A_prng.read(), A_prng.read(),
     A_prng.read(), A_prng.read(),  A_prng.read()]
```

Denote `A_prng`'s outputs as `a0`, `a1`, `a2`, etc.; ditto for `B`.

We have:
* `x = [a1, a2, a3, b0, b1, b2]`
* `y = [a1, a2, a3, a4, a5, a6]`

Do the values in `x` exhibit unintended correlations, but `y`'s values do not?  How is that so?

I agree that in pydrake we should better advise users how to properly seed things (no `/dev/urandom`!), but I don't yet see why we need to share a single simulation-wide `pydrake.common.RandomGenerator` object across_all objects that require random reads.

---

[1] I'm using "read" to denote drawing a 32-bit output integer from the PRNG, even though it's variously `operator()`, `__call__`, `__index__` etc. in various languages and situations.

RussTedrake, 2020-01-27:
Very nice.  Thanks!
The case that I was concerned with is
```
prng = std::mt19337(fixed_seed)
A_prng = numpy.random.RandomState(seed=prng.read()) 
A_prng.read() // discard one
B_prng = numpy.random.RandomState(seed=prng.read()) 
x = [A_prng.read(), A_prng.read(), A_prng.read()]
y = [B_prng.read(), B_prng.read(), B_prng.read()]
```
possibly generating `x == y`.  If the "internal state is not the seed" is sufficient to imply this is not a concern, then great, and we should just document that `np.random.RandomState(generator())` is the recommended usage.

I do also think we should recommend how to produce random runs using `/dev/urandom` or `time`.

jwnimmer-tri, 2020-04-02:
After writing the above, I did a lot more reading about properly seeding monte-carlo simulations.  There is a whole body of work on the topic.  I think the notes above are "good enough" and avoid the terrible problems you were worried about, but we could probably still do better in terms of using more of the generator's potential state space during initialization (the seed bits are much smaller than the state bits) -- possibly by foregoing the mersenne twister family entirely.

PR #14102 (2021-03-26): python: document RandomGenerator use with numpy
Resolves #12632

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/14102)
commit c8be496e (2021-03-26): pydrake: Document RandomGenerator use with numpy (#14102)

Resolves #12632
closed 2021-03-26 by jwnimmer-tri