RobotLocomotion/drake #12633: pybind fails to convert an ndarray of shape (0, 2) to eigen?

RussTedrake, 2020-01-26:
The symptom is truly strage.  This code
```
import numpy as np
from pydrake.systems.primitives import LinearSystem

A=np.zeros((0,0))
B=np.zeros((0,2))
C=np.zeros((2,0))
D=np.eye(2)
sys = LinearSystem(A=A,B=B,C=C,D=D)
```
fails with 
```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-111-9166b2860e4e> in <module>()
      6 C=np.zeros((2,0))
      7 D=np.eye(2)
----> 8 sys = LinearSystem(A=A,B=B,C=C,D=D)

TypeError: __init__(): incompatible constructor arguments. The following argument types are supported:
    1. pydrake.systems.primitives.LinearSystem_[float](A: numpy.ndarray[float64[m, n], flags.f_contiguous], B: numpy.ndarray[float64[m, n], flags.f_contiguous], C: numpy.ndarray[float64[m, n], flags.f_contiguous], D: numpy.ndarray[float64[m, n], flags.f_contiguous], time_period: float = 0.0)

Invoked with: kwargs: A=array([], shape=(0, 0), dtype=float64), B=array([], shape=(0, 2), dtype=float64), C=array([], shape=(2, 0), dtype=float64), D=array([[ 1.,  0.],
       [ 0.,  1.]])
```

But this works...
```
import numpy as np
from pydrake.systems.primitives import LinearSystem

A=np.zeros((0,0))
B=np.zeros((0,1))
C=np.zeros((1,0))
D=np.eye(1)
sys = LinearSystem(A=A,B=B,C=C,D=D)
```
Similarly, taking the original code, but using `B = np.zeros((0,1))` or `B=np.zeros(1,2)` results in the expected behavior -- it calls to the bound constructor and then complains about the size mismatch.

`B=np.zeros(0,3)` or higher also fail unexpectedly.

I'm stumped.

EricCousineau-TRI, 2020-01-27:
Seems like this is due to the usage of `const Eigen::Ref<const MatrixXd>&` for the input arguments.
With just `const MatrixXd&`, there's no error:
https://github.com/EricCousineau-TRI/drake/commit/75d7c21d1665e0af2bfcd1e392a5ea1b0beb7636

Will see if it's due to our fork, or something different.

RussTedrake, 2020-01-27:
Thanks!  Just to confirm, we should be able to leave the `Eigen::Ref` in place, correct?  It seems like a clear bug that it would match on all of the variables except `B`, and match `B` in some occurrences, but not others.

EricCousineau-TRI, 2020-01-27:
I've filed the upstream issue here: https://github.com/pybind/pybind11/issues/2098
Add PR to fix it here: https://github.com/pybind/pybind11/issues/2098

Basically, it's just some overly conservative stride-checking. It was ignored for the 1xN / Nx1 case, and just needs a slight adjustment for the 0xN / Nx1 case. Once we have this patch in our fork, this code should work.

PR #12641 (2020-01-28): workspace: Update pybind11 fork to latest commit
Resolves #12633

Requires:
- [x] https://github.com/RobotLocomotion/pybind11/pull/38

FYI @RussTedrake

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12641)
closed 2020-01-28 by EricCousineau-TRI