RobotLocomotion/drake #12634: broken link to SOS tutorial from mathematical_program.ipynb

RussTedrake, 2020-01-27:
It looks like the SOS tutorial link is pointing to @hongkai-dai 's google drive.  When I click on it from [binder](https://mybinder.org/v2/gh/RobotLocomotion/drake/master?filepath=tutorials/mathematical_program.ipynb)
 -- it's at the very bottom of the page -- it looks to try to open the colab page but renders very very oddly.

I think the right solution is to land the SOS tutorial in `drake/tutorials`.

EricCousineau-TRI, 2020-01-27:
@hongkai-dai Would you like me to port the colab page to Drake? We can always go back and do a cleanup pass afterwards?

hongkai-dai, 2020-01-27:
@EricCousineau-TRI , yes that would be great. I will do a clean up pass later after you port the tutorial to Drake.

PR #12642 (2020-01-30): tutorials: Import SOS tutorial from colab
Resolves #12634

This is just a quick import.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12642)
closed 2020-01-30 by EricCousineau-TRI