RobotLocomotion/drake #12638: Stale docs for Drake GitHub issue labels

jwnimmer-tri, 2020-01-27:
@RussTedrake it looks like you (or someone else at MIT) added more `team: ` names to Drake's list of issue labels in GitHub?  That's fine, but then also we need updates to:
- the textual enumeration of teams and owners at https://drake.mit.edu/issues.html, and
- the query filter in https://drake.mit.edu/platform_reviewer_checklist.html hyperlink.

jwnimmer-tri, 2020-01-27:
(Alternatively, keep only `team: robot locomotion group` and add a different label, e.g., just `6.832` to flag them as related to courses.)

RussTedrake, 2020-01-27:
Apologies; did not appreciate the dependency.  

I like that last suggestion (only team: robot locomotion group, then extra tags).  Will implement it tonight if nobody beats me to it.

RussTedrake, 2020-01-27:
Changed `team: MIT 6.881` => `type: MIT 6.881` and `team: MIT 6.832` to `type: MIT 6.832`, and added `team: robot locomotion group` to any issues that would have been left without a `team` assigned.  Will PR the update to the docs now.

PR #12639 (2020-01-27): Update links to reflect modified team names. 
MIT teams have been consolidated to `team: robot locomotion group` and `MIT 6.832` and `MIT 6.881` are separate (non-team) tags.

Resolves #12638

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12639)
commit ef7ffbf5 (2020-01-27): Update links to reflect modified team names.  MIT teams have been consolidated to `team: robot locomotion group` and `MIT 6.832` and `MIT 6.881` are separate (non-team) tags. (#12639)

Resolves #12638
closed 2020-01-27 by jwnimmer-tri