RobotLocomotion/drake #12645: tutorials: Need example of showing meshcat on Binder (if possible)

EricCousineau-TRI, 2020-01-28:
> Before you post:

> * Have you reviewed the documentation on
[Getting Help](https://drake.mit.edu/getting_help.html)?
> * Have you reviewed existing GitHub issues and StackOverflow posts?
> * Have you rerun `install_prereqs.sh` on your current commit of `drake`?

> If you have encountered a bug (installation, segfaults, etc.), please try to
post reproduction steps and as much relevant output as possible. Consider
posting long snippets as a [Gist](https://gist.github.com/).

EricCousineau-TRI, 2020-01-29:
For reference, it seems like `pyntcloud` has notebooks that run on Binder and can show (web3d?) visualization widgets:
https://github.com/daavoo/pyntcloud/blob/master/README.rst

Example Binder link:
https://mybinder.org/v2/gh/daavoo/pyntcloud/master?filepath=examples/[visualization]%20Polylines.ipynb

EricCousineau-TRI, 2020-01-29:
Just to expand, this difference in functionality can be reproduced in a local Docker container, per this bit:
https://github.com/RobotLocomotion/drake/tree/1ffb58bd1fa5e0c91e38c4bc7ec0de4cbcbfa56e/.binder#docker-image-for-binder
I cannot see the `meshcat` visualization in Jupyter within docker, but can see it on my local machine.

I can also install `open3d==0.9.0` (with a newer version of Jupyter notebook), and can visualize point clouds with code like:
```
# http://www.open3d.org/docs/release/tutorial/Basic/working_with_numpy.html
import numpy as np
import open3d as o3d

# generate some neat n times 3 matrix using a variant of sync function
x = np.linspace(-3, 3, 401)
mesh_x, mesh_y = np.meshgrid(x, x)
z = np.sinc((np.power(mesh_x, 2) + np.power(mesh_y, 2)))
z_norm = (z - z.min()) / (z.max() - z.min())
xyz = np.zeros((np.size(mesh_x), 3))
xyz[:, 0] = np.reshape(mesh_x, -1)
xyz[:, 1] = np.reshape(mesh_y, -1)
xyz[:, 2] = np.reshape(z_norm, -1)

# Pass xyz to Open3D.o3d.geometry.PointCloud and visualize
pcd = o3d.geometry.PointCloud()
pcd.points = o3d.utility.Vector3dVector(xyz)

# http://www.open3d.org/docs/release/tutorial/Basic/jupyter.html
visualizer = JVisualizer()
visualizer.add_geometry(pcd)
visualizer.show()
```

jamiesnape, 2020-02-25:
The first part of this is going to be whether we can actually expose a port when running on Binder. If it happens to run the equivalent of `docker run -P` (it uses kubernetes, I think, however) then that part is easy. Locally, you would want to add something like `-p 7000:7000` to the `docker run` command.

jamiesnape, 2020-02-25:
And as you would imagine, it does not allow you to expose 7000, so no websocket connection, so no meshcat visualization.

jamiesnape, 2020-02-26:
(so the short answer would be that using a different visualizer would be a million times easier)

EricCousineau-TRI, 2020-02-26:
Aye, makes sense...

jamiesnape, 2020-02-26:
Are going to therefore close this as not possible or re-scope it as use a different visualizer or rewrite much of meshcat?

EricCousineau-TRI, 2020-02-26:
Re-scoping to achieve some level of 3D visualization on Binder.

EricCousineau-TRI, 2020-02-26:
Also filed: https://github.com/rdeits/meshcat/issues/67

EricCousineau-TRI, 2020-02-27:
Next step is to try another `three.js`-based solution, but one that doesn't require WebSockets (may be slower... but works?)

jamiesnape, 2020-02-27:
We also have something like `itkwidgets` based on vtk.js:

https://pypi.org/project/itkwidgets/

with Binder examples here:

https://mybinder.org/v2/gh/InsightSoftwareConsortium/itkwidgets/master?urlpath=lab/tree/examples

BetsyMcPhail, 2020-04-30:
I did some experimentation with itkwidgets.

I tried some of the itk examples linked above (#13182).  These examples work in Binder assuming itk and itkwidgets are available.Note that the version 0.27.0 of itkwidgets is not rendering in Binder ([itkwidgets issue](url)), I used itkwidgets==0.26.1 to test. 

As a sidenote, I tried pip installing itk and itkwidgets directly from Binder , e.g.

```
import sys
!{sys.executable} -m pip install itk
!{sys.executable} -m pip install itkwidgets

```
But I couldn't import itkwidgets?

The vtk examples currently don't work in Binder ([brief explanation](https://github.com/InsightSoftwareConsortium/itkwidgets/issues/230#issuecomment-568299114)).

I haven't explored this at all, but this [disscussion ](https://discourse.vtk.org/t/python-vtk-for-visualization-in-a-jupyter-notebook-running-on-docker/622 )seems to indicate that it might be possible?

EricCousineau-TRI, 2020-05-07:
@BetsyMcPhail For your WIP branch, can you instead try to hack the `.binder/Dockerfile` (not mac prereqs) to install a custom-built version of VTK for the version you need and then post the exact command-line you used to run the Docker container image locally?

EricCousineau-TRI, 2020-05-07:
If it takes too long for `vtk.js`-based solutions, then we may want to just do direct `pythreejs` / `three.js` solutions as shown in above examples.

BetsyMcPhail, 2020-05-14:
#13182 has been updated with a new notebook (_tutorials/3dvisualization_tests_) that has working examples from itkwidgets, open3d and pyntcloud.

`.binder/Dockerfile` has been updated to install and setup any necessary modules.

Use the [docker image for Binder](https://github.com/RobotLocomotion/drake/tree/1ffb58bd1fa5e0c91e38c4bc7ec0de4cbcbfa56e/.binder#docker-image-for-binder) to try locally.

ToffeeAlbina-TRI, 2020-05-14:
@EricCousineau-TRI do you want to evaluate this as is and decide whether/what other visualizer options Betsy should explore?

EricCousineau-TRI, 2020-05-21:
Thanks! I've just ran through the examples and provided some feedback in the PR.

EricCousineau-TRI, 2020-05-21:
@jamiesnape When you have a chance, per your prior comment: 

> And (after just trying) as you would imagine, it does not allow you to expose 7000, so no websocket connection, so no meshcat visualization.

Would you or Betsy have time to add a permalink link to the JupyterHub code which indicates that this is not possible?

jamiesnape, 2020-05-21:
There isn't public code. It would be a server firewall configuration.

EricCousineau-TRI, 2020-05-22:
Sorry; what I meant was if there's anything in JupyterHub's deployment configuration that might preclude additional ports. I understand that won't eliminate the possibility that Binder's hosts may place additional restrictions.

EricCousineau-TRI, 2020-05-22:
And also, it would be nice to confirm beyond assumption.
Can you see if a similar question has been asked / answered in the [Binder forum](https://discourse.jupyter.org/c/binder), or in the [Binder docs](https://mybinder.readthedocs.io/en/latest/)?

EricCousineau-TRI, 2020-05-22:
For example, it looks like you can run a Bokeh server through a proxy?
https://mybinder.readthedocs.io/en/latest/sample_repos.html?highlight=port#running-a-bokeh-server-with-binder

jamiesnape, 2020-05-22:
> For example, it looks like you can run a Bokeh server through a proxy?
> 
> https://mybinder.readthedocs.io/en/latest/sample_repos.html?highlight=port#running-a-bokeh-server-with-binder

The proxy looks interesting, though it would still need some moderate patching on the meshcat side if you can indeed proxy the server through tcp/8888 and forward that to tcp/443.

EricCousineau-TRI, 2020-05-22:
Sweet! Moderate patching sounds wayyy better than re-architecting, esp. if the proxying gets us what we want!

Is it possible to queue that up to attempt with a Meshcat branch, say over the next month?

---

FTR, @rdeits was able to take some time to comment on this issue:
https://github.com/rdeits/meshcat/issues/67

@BetsyMcPhail On a separate thread, would it be possible to try Robin's suggestion of transmitting static HTML, first for a static scene, then for an animation?
Similar to [this PR comment](https://github.com/RobotLocomotion/drake/pull/13182#pullrequestreview-416324645), could you make it look like this workflow?
https://nbviewer.jupyter.org/github/RobotLocomotion/drake/blob/nightly-release/tutorials/pyplot_animation_multibody_plant.ipynb#Run-with-Playback

(My estimate is this would take ~3 Perfect Engineering Hours - is this viable within the next 2 weeks?)

At present, no need to patch the current `drake.systems.meshcat_visualizer` code; just writing a targeted Meshcat-only example would work to boot!

BetsyMcPhail, 2020-05-22:
> @BetsyMcPhail On a separate thread, would it be possible to try Robin's suggestion of transmitting static HTML, first for a static scene, then for an animation?
> Similar to [this PR [...]

Sounds like a good plan - within the next 2 weeks should be doable.

EricCousineau-TRI, 2020-05-28:
Per convo with @ToffeeAlbina-TRI, I've updated the overview, enumerating some of the potential solutions discussed here.

At present, if we can do port-forwarding on Binder, then `meshcat` may be the lowest-energy solution.
Alternatively, we can static HTML rendering, recording the animation (like `PyPlotVisualizer`).
Failing that, then making tooling around `itkwidgets.js` / `vtk.js` / `three.js` directly may be the main viable solution (but higher overhead).

EricCousineau-TRI, 2020-06-04:
Just added some checkboxes at top-level. Can sequence it out if need be!

EricCousineau-TRI, 2020-06-18:
Re-assigning to Jamie to investigate the port proxy on Binder.

jamiesnape, 2020-06-23:
I have looked deeply into why yet, but it is not looking promising with the proxy. Certainly the HTTP is being correctly proxied from 7000 to 8888, but the upgrading to a websocket is running into problems. I will keep working on it, but I am not hopeful.

RussTedrake, 2020-06-24:
I've been thinking about this again.  A few observations:
- I believe we consider the actual Binder service to be flakey (probably too flakey for use with student problem sets, for instance).  I think the work we are doing here is really paving the way to us having 3d visualization in a JupyterHub / BinderHub if we have our own instances.  Correct?
- Colab was a reliable resource for class.  I was less optimistic about getting visualization working in Colab, but have been experimenting a bit more this morning.  
  - I confirmed that [Altair works, even interactively in colab](https://colab.research.google.com/drive/1r03r_k1d_kiCm-j_gNIRU2Wi_XIYDSWB?usp=sharing).
  - I've also found some examples of three.js (which meshcat is built on) [displaying in colab](https://colab.research.google.com/drive/1nrK1bg8rqoYGVZfd4cuB3ndFvuUafVuu#scrollTo=y-D3CG9P-Nl1).    So I think that the issues with meshcat are in principle surmountable; it's the network connection between the client and server that is causing problems, not the actual visualization?

jamiesnape, 2020-06-24:
> I've been thinking about this again. A few observations:
>
> * I believe we consider the actual Binder service to be flakey (probably too flakey for use with student problem sets, for instance).  I [...]

We would have more flexibility with our own JupyterHub, so the solution could be different/easier.

> * Colab was a reliable resource for class.  I was less optimistic about getting visualization working in Colab, but have been experimenting a bit more this morning.
>
>   * I confirmed that [Altair [...]

Yes, it is the network connection. In principle, most of the JavaScript visualizers should work fine with any of the managed platforms so long as we can get the necessary data to them.

RussTedrake, 2020-06-24:
Poking around a little more, it looks like `meshcat-server` is not actually needed.  I'm new to this, but based on some code browsing, I *think* we could probably hand the meshcat viewer js directly to the jupyter notebook (instead of the iframe through the meshcat-server), and connect to it via a websocket from our pydrake visualizer.  And a little googling suggests that websockets do (must?) work on colab.  ?

@rdeits -- does that sound viable to you?

RussTedrake, 2020-06-25:
[Here is a link ](https://colab.research.google.com/drive/1O-WgLtqKGuHKEeD4CFT69870dPEJRsQJ?usp=sharing) to my experiments with meshcat on colab.

RussTedrake, 2020-06-25:
ok, I've got it basically working in the notebook above.  `ngrok` for the win.
(specifically, I can launch an meshcat-server on the colab notebook, connect to it on colab with meshcat-python, and open up a new window on my local machine and see the results).  the last step is to get the jupyter_cell rendering working right in the colab notebook, which is close.

RussTedrake, 2020-06-28:
@jamiesnape -- i've got a concrete ask.  I've cleaned up my [colab example](https://colab.research.google.com/drive/1O-WgLtqKGuHKEeD4CFT69870dPEJRsQJ#scrollTo=wgdiSo0NTwQJ), which describes the last hurdle: I need `wss` to work through `ngrok` in order to connect to meshcat in a colab cell.  Any help is appreciated.

jamiesnape, 2020-06-30:
I will look into it.

jamiesnape, 2020-06-30:
A small amount of progress. This gets me a connection, but not necessarily a working connection:
```
from IPython.display import HTML

HTML("""
<div id="meshcat-pane" style="height: 400px; width: 100%; overflow-x: auto; overflow-y: hidden; resize: both">
</div>

<script type="text/javascript" src="{js_url}"></script>
<script>
    var viewer = new MeshCat.Viewer(document.getElementById("meshcat-pane"));
    viewer.connect({wss_url});
</script>

<script id="embedded-json"></script>
""".format(js_url='https://localhost:7000/static/main.min.js', wss_url='`wss://${location.host}:7000`'))
```
Among other things seems `wss://localhost:7000` does not work the same way as `https://localhost:7000` in relation to a server running on Colab. Also `location.host` is missing the port number (so is hard-coded for the moment).

RussTedrake, 2020-07-17:
Spent another hour on this today w @sjahl (thanks!).  Here is my current understanding.  tl;dr: I think I need to teach meshcat server to accept a secure websocket.

I agree with @jamiesnape -- and think we shouldn't need ngrok.  Even if we wanted to use ngrok, opening a TLS tunnel on ngrok requires a paid version.  I'm not against paying, but do not have an option that would allow *any* user of the notebook to have a paid ngrok version.  So the ngrok branch of thinking is pruned.

I believe that the problem now is that the meshcat server is not expecting a wss conneection.  This is in my power to fix.
- [Meshcat is using tornado](https://github.com/rdeits/meshcat-python/blob/master/src/meshcat/servers/zmqserver.py).  
- Tornado has some details about [how to setup wss here](https://www.tornadoweb.org/en/stable/websocket.html#tornado.websocket.WebSocketHandler), including some potential hurdles to jump through (like opening a normal html page in order to handle an "accept this cert" popup.
- I would love to generate a generic certificate once, but we suspect that the certificate will need to be hostname specific.  That implies that each colab notebook will need to create a certificate and pass it into the startup for the meshcat server.  [Here's a doc for creating the certificate](https://devcenter.heroku.com/articles/ssl-certificate-self).

RussTedrake, 2020-07-18:
Update: I got a lot farther, but got stuck again.  [notebook is here](https://colab.research.google.com/drive/1wOtqXes5155_o2AShyfubK9VZcWWJVpQ?usp=sharing).
- was able to generate a self-signed cert
- was able to pass those arguments into meshcat server and have it launch using the certs... as verified by `curl`.  But `curl` needs `--insecure` to accept the self-signed cert.
- i can't use the standard meshcat trick of getting the meshcat src from the https request (since I can't make an iframe accept the self-signed certificate), but I got around this by just writing the .js file into my frame directly.
- current blocker: i still can't get my websocket connection back to my (now self-signed) wss port. I don't know how to tell it to accept that certificate.   I've got an even simpler websocket example in the notebook for testing.  Posts like [this one](https://stackoverflow.com/a/27940375) suggest that one should browse to a plane html url first, so that the browser prompts me to accept the certificate, but I've been unable to get that to work, despite trying it from a bunch of angles.  Will have to shelve this again for now.

RussTedrake, 2020-07-22:
Minor update:  I've got meshcat websockets working over SSL on my local machine (not colab), and only with the visualizer in a separate window, not a jupyter cell.  As expected, it results in an error message in chrome, until you manually accept the self-signed certificate.  https://github.com/rdeits/meshcat/pull/71 

Next step (after landing the meshcat PRs):
- connect via ssl in a jupyter cell on my local machine.

RussTedrake, 2020-07-22:
Huzzah.  I've got meshcat running in my local notebook via viz.  Once my meshcat PRs land, the following works:

cell 1:
```
import meshcat

zmq_url="tcp://127.0.0.1:6000"
vis = meshcat.Visualizer(zmq_url=zmq_url)

box = meshcat.geometry.Box([0.5, 0.5, 0.5])
vis.set_object(box)

vis.jupyter_cell()
```

cell 2:
```
import math
import time
for i in range(500):
    theta = (i + 1) / 100 * 2 * math.pi
    now = time.time()
    vis.set_transform(meshcat.transformations.rotation_matrix(theta, [0, 0, 1]))
    time.sleep(0.01)
```
when I launch meshcat server via:
```
python3 -m meshcat.servers.zmqserver --certfile localhost.crt --keyfile localhost.key
```
(and have accepted the certificates at least once).  Everything flows through ssl.

EricCousineau-TRI, 2020-07-30:
Just for now, will re-assign this to you Russ until there're more action items for someone else.

RussTedrake, 2020-08-01:
Just to update my thoughts on this...

After talking to @rdeits a week or so ago, he mentioned that the entire meshcat network setup was actually inspired by the way that jupyter kernels work (websockets and zmq).  Assuming colab is the same, it might be that the most viable option here is actually to figure out if we can piggyback on the existing websocket connection between the browser and the server instead of trying to instantiate our own.  @rdeits said he had managed to do that with Jupyter once before.

duburcqa, 2020-08-04:
Hi guys ! I had a similar issue. I managed to get it work after "fixing" websocket, by enabling cross-traffic. Here is a working snippet enabling to render a custom javascript meshcat viewer in jupyter:
```
from IPython.display import HTML

HTML("""
<div style="height: 400px; width: 100%; overflow-x: auto; overflow-y: hidden; resize: both">
    <iframe id="testiframe" src='data:text/html,
        <div id="meshcat-pane" style="height: 400px; width: 100%; overflow-x: auto; overflow-y: hidden; resize: both">
        </div>

        <script type="text/javascript" src="http://127.0.0.1:32784/static/main.min.js"></script>
        <script>
            var viewer = new MeshCat.Viewer(document.getElementById("meshcat-pane"));
            viewer.connect("ws://127.0.0.1:32784");
        </script>

        <script id="embedded-json"></script>'; style="width: 100%; height: 100%; border: none">
    </iframe>
</div>
""")
```

rdeits, 2020-08-04:
Ok, here's a direction which doesn't require any port forwarding, server configuration, certificates, or proxy. Rather than using a websocket, we can send commands from python to the frontend using the built-in Jupyter comms protocol. I've set up a dumb example that renders an iframe and then sends messages to it from Python here: https://github.com/rdeits/meshcat-python/blob/jupyter-comms-demo/comms.ipynb

This means:
* no websocket (other than what Jupyter already uses under the hood)
* no proxy
* no port forwarding
* no security holes (we're just using the comms that the notebook already provides)
* no certificates
* no ZMQ server at all

This would require some changes to meshcat to support sending messages this way instead of over ZMQ, but it might not be too bad. Essentially, we'd have to replace the calls to `zeromq.send` with `command_channel.send()`. Figuring out a nice way to do that without breaking the existing use-cases would require some thought, but I don't think it's too bad. 

 @RussTedrake this is what I was referring to when I talked about using the Jupyter comms protocol itself. It should also work on Colab (although I haven't tested) since it's the same underlying mechanism that things like altair and ipywidgets use.

RussTedrake, 2020-08-05:
@duburcqa -- thanks.  That one we knew how to do, and Robin already has it basically available in the `jupyter_cell()` method.  The trick is that colab won't allow the connection via ws.  It needs wss.

@rdeits -- awesome!  it worked exactly as expected on my local machine, but my first copy/paste to colab did not:
https://colab.research.google.com/drive/1wsgE6EzRORqc2uXotRaL5lpbCZYvBYvH?usp=sharing
It looks like `window.parent.Jupyter` is `undefined` on colab.  I found `window.colab` (also `window.parent.colab`), which has `kernel`, but not `comm_manager`.  Will dig a little more.

RussTedrake, 2020-08-05:
Ah, just found this: https://github.com/googlecolab/colabtools/blob/master/packages/outputframe/lib/index.d.ts
(from https://colab.research.google.com/notebooks/snippets/advanced_outputs.ipynb#scrollTo=Ytn7tY-C9U0T)

RussTedrake, 2020-08-05:
Made some progress, but hit (I think) a missing feature/export in colab.  [I've asked on stackoverflow](https://stackoverflow.com/questions/63263921/is-there-a-way-to-register-a-message-handler-callback-on-a-google-colab-kernel-c).

RussTedrake, 2020-08-05:
OK, I have a proof of life, albeit with the probably inefficient version of opening a new comm channel for every message.
https://colab.research.google.com/drive/1O-WgLtqKGuHKEeD4CFT69870dPEJRsQJ
As @rdeits says, I think the rest is just plumbing, and shouldn't be that bad!

RussTedrake, 2020-08-05:
Woot!  I think I've get everything working now (though it will get more efficient if the stackoverflow generates a response).  
https://colab.research.google.com/drive/1O-WgLtqKGuHKEeD4CFT69870dPEJRsQJ
meshcat PR is linked above.

jcarpent, 2021-08-24:
@RussTedrake We would also like to integrate Pinocchio within Colab/Binder with Meshcat viewer. Did you manage to integrate the solution into Drake framework?
@duburcqa Did you manage to also provide this portage within your framework?

RussTedrake, 2021-08-25:
@jcarpent -- see also #13038, which is now almost complete.  There are a lot of details and moving part that I could help you understand if you like.
Long story short:
- I've been using Meshcat successfully on Colab for a long time now; it can work quite well.  This remaining issue was only about embedded colab into the actual notebook as a iframe.  I have not gotten it working, mostly because most people actually prefer to have meshcat in a separate window anyhow.
- The primary challenge of putting meshcat into an iframe in the notebook is the combination of requirements that the notebook iframes require https/wss sources, but I currently use ngrok to tunnel http/ws out from the server.  I've added wss support into meshcat, but the free ngrok won't support https/wss and their pricing plans don't make sense for this cloud workflow.  If you hosted your own tunnel you could get around it.
- I'm actually now likely transitioning away from Colab.  It's very generous in terms of compute resources, but it is not a stable environment to support.  (the libraries and even python versions can change out from under you with no notice and no recourse).  https://github.com/RussTedrake/manipulation/issues/144 
- I've implemented the versions of meshcat that work only through kernel comms, but Colab / Deepnote / etc all offer their own implementations (if at all) of the kernel comms, so writing one clean meshcat implementation that works for all of them is likely impossible.

jcarpent, 2021-08-25:
Dear @RussTedrake,

Thanks a lot for the clear and detailed answer.
Having your feedback on Colab will for sure help us to not stuck into the same local minima.
I did not know about Deepnote, I will have a look into it.

I'm also following #13038: this is a very interesting feature that may be useful to a lot of people. 
Could we imagine transferring it into an independent package `meshcat-cpp`? I may give some help on that and I know for instance @traversaro is also motivated.

Thanks again @RussTedrake for your very helpful feedback.

cc: @nim65s @proyan @manifoldFR

RussTedrake, 2021-08-25:
I definitely considered `meshcat-cpp`, but decided against in the first pass to streamline things.  It currently depends on Drake's geometry types and rigid transform type, (plus build system / CI / helpers for memory/exception management, etc), as I didn't want to have to reinvent them for a more general implementation.

jcarpent, 2021-08-25:
Perfect. I totally understand.
Do not hesitate to ping us if you need support.

nim65s, 2021-09-01:
Hello,

I tried to embed meshcat in a jupyter cell on binder: https://github.com/nim65s/binder-meshcat-pinocchio-test.
You can try it at: https://mybinder.org/v2/gh/nim65s/binder-meshcat-pinocchio-test/main
It looks like it is working, but this probably need more tests, especially about speed.

It is mostly a copy-paste from https://github.com/rdeits/meshcat-python/pull/74, with the following changes:
1. removed the iframe. I'm not sure why it was needed in the first place. Did I miss something ? We can put it back if needed.
2. split meshcat inclusion into `HTML()` & `Javascript()`
3. host meshcat somewhere in HTTPS with `Access-Control-Allow-Origin: *`
4. use requirejs to get it (requirejs is already embedded in jupyter notebooks)
5. initialize communication from the client instead of the server (I was trying to ensure that if the communication is not properly established, the python part will notice)
6. removed parts related to Google Colab
7. instead of sending all the meshes by the communication bus, I hosted them somewhere in HTTPS, and I'm sending only an url. After that, they are fetched on the client side: https://github.com/nim65s/meshcat/commit/db88112a0046771fd5231d71e9978f0c11bff458

A few comments about that:
- https://github.com/duburcqa/jiminy/issues/403 raised rate limit issues. I also managed to get them locally, so I implemented (3) & (7) to help with that.
- For this test, I'm using a small personal server for (3) & (7), but I think we should consider using a CDN for this kind of things, at least for (3). Anyway, this server is not a permanent one, it's just here for a few tests
- (6) can obviously be reverted, if we want this thing to work on multiple jupyter notebook providers. I didn't read the whole thread, but I guess there are stronger issues with it.
- my tests are tied to pinocchio and its MeshcatViewer abstraction, but we can easily extract the more generic parts

I'd be glad to discuss a bit about this, before going further and opening PRs ; so if you have any comment or subjection, please don't hesitate :)

jwnimmer-tri, 2022-04-20:
For reference / archival or if anyone is interested --

Here's some code that I've used in Deepnote.  It automatically opens up a Meshcat display in a new browser tab.

```python
import meshcat
import os
from IPython.display import HTML, Javascript, display

with open(os.path.dirname(meshcat.__file__) + '/viewer/dist/main.min.js', 'r') as f:
    main_min_js = f.read()

def js_esc(data):
    bs = '\\'
    return data.replace(bs, bs*2).replace('\n', bs + '\n').replace("'", bs + "'")

body = f"""
<div id="meshcat-pane" style="height: 100%; width: 100%; overflow-x: auto; overflow-y: hidden; resize: both"></div>
"""
title = f"""
MeshCat
"""
mine = """
document.meshcatViewer = new MeshCat.Viewer(document.getElementById("meshcat-pane"));
"""
js = main_min_js + ";" + mine
escaped_body = body.replace("\n", "\\n")
js = f"""
var newWindow = window.open("", "meshcat");
newWindow.document.title = '{js_esc(title)}';
newWindow.document.body.innerHTML = '{js_esc(body)}';
var script = newWindow.document.createElement('script');
script.text = '{js_esc(js)}';
newWindow.document.head.appendChild(script);
window.document.meshcatViewer = newWindow.document.meshcatViewer;
console.log("MeshCat is ready!");
"""
display(Javascript(js))
```

This allows for opening the window without needing an https server for the HTML code.  It also allows code in notebook cells to inject Javascript code into the meshcat tab.

It does *not* resolve any problems related to the meshcat websocket messages; those would still need to come in via a wss socket.

closed 2022-08-29 by RussTedrake

vincekurtz, 2026-05-18:
For anyone who might still be looking at this issue for a solution, these days it's easy:

```
!pip install drake

from google.colab import output
from pydrake.all import StartMeshcat

meshcat = StartMeshcat()
output.serve_kernel_port_as_iframe(meshcat.port())
```