RobotLocomotion/drake #12648: Need to be able to define cache entries that start out disabled (for debugging).

sherm1, 2020-01-29:
When defining cache entries (or output ports, which implicitly define cache entries) it is possible to make errors declaring their dependencies, or elsewhere in the code that calculates results. One way to catch that is to disable caching altogether (forcing recalculations) and see if the results change (they should not). If the results do change it can then be very difficult in a big Diagram to find the guilty party.

It would be helpful in that case to disable just a single cache entry at a time to determine which one is causing trouble. The cache system already has that feature but currently it can only be invoked after a Context is available, assuming you can find the relevant Context and the CacheEntryValue object within that corresponds to a particular declared CacheEntry (or OutputPort). It would be much easier to mark the CacheEntry at time of declaration so that every Context it creates comes up with caching disabled for that CacheEntry. I used a hacked version of that to debug #12643 -- we should add a real version to master.

Something like `cache_entry.set_is_disabled_by_default(bool)` is all that's needed for explicitly declared cache entries. Somewhat trickier for output ports since the returned ports are const. Consider adding an optional constructor flag to create a LeafOutputPorts with a default-disabled cache entry?

amcastro-tri, 2020-01-29:
Some visualization tool showing "hot spots" as they get updated during the complex logic of `AdvanceTo()` would be amazing. I wonder if our diagram viz tools could be extended somehow to this end?

jwnimmer-tri, 2020-01-31:
If it's for debugging anyway, maybe just use a singleton, instead of polluting all of the other APIs?
```
drake::framework::CacheEntry::SetGloballyDisabled(const std::string& entry_name);
```

Where `entry_name` is the value of `CacheEntry::FormatName()`, which we could trace-level log as they are created (so the user knows what valid values can be).

closed 2020-03-04 by soonho-tri