RobotLocomotion/drake #12652: sytems::Diagram has no API call for allocating anything other than discrete variables

rcory, 2020-01-30:
Since `systems::diagram::AllocateDiscreteVariables()` is already provided by the API, it should also provide `AllocateAbstractVariables()`, `AllocateContinuousVariables()`, etc.

sherm1, 2020-01-30:
Related: #12616

jwnimmer-tri, 2020-01-30:
Continuous variables are `AllocateTimeDerivatives` already, available in `System`.

Note that I will be deprecating `LeafSystem::AllocateContinuousState` in #12630 or before.

I suppose for abstract we should have an API -- the workaround in the meantime is `AllocateContext` + `get_abstract_state`?  But what is the use case where you do not want to use `AllocateContext` instead?  Most things deal with the whole context, not just a portion of it.

I do worry about advertising APIs for these though before #9171 is finished.

sherm1, 2020-06-01:
@rcory in view of @jwnimmer-tri's comment above do you think these are still needed?

jwnimmer-tri, 2020-12-09:
Closing for lack of response.

closed 2020-12-09 by jwnimmer-tri