RobotLocomotion/drake #12655: sphinx documentation for pydrake is missing overview docs

RussTedrake, 2020-01-31:
in discussing with some users about why the python documentation is not working well for them, many complain that our sphinx output is just a list of methods, with no apparent organization.  I think that publishing the doxygen pages (e.g. https://drake.mit.edu/doxygen_cxx/index.html#://) to the front of the sphinx pages would go a long way, even if it temporarily has some smatterings of c++ documentation in it like the rest of the pydrake docs.

In the very short term, I think folks would get a lot out of even having a trivial front page for sphinx that points users to the doxygen c++ (and explains that doxygen c++ is the first class citizen, they may want to browse there first).

cc @EricCousineau-TRI

EricCousineau-TRI, 2020-02-03:
I'm starting this off; will submit a PR with high-level overview, and will then update each module.

EricCousineau-TRI, 2020-02-03:
Some initial notes:

* Trying to refer to a top-level Doxygen API page is inconsistent. Some components / modules have good module pages, others... not so much. Also, namespace division is also inconsistent, and namespaces do not have general overview.
* Referring to directories may be the better approach; however: (a) our current Doxygen generation does not produce an overview of classes / functions / modules at the directory level, only at the file level, and (b) the links for directories are based on what looks like a hash that I'm guessing is from an index (and thus hard to generate independently, i.e. only via Sphinx).

\cc @sherm1 along the topic of C++ doc organization

sherm1, 2020-02-03:
@EricCousineau-TRI I'm thinking this might be best tackled one module at a time rather than as a generic reorg of the C++ docs. What constitutes a meaningful module is subjective as reflected by the odd assortment of topics on the Doxygen front page. Maybe better to pick an important topic, make sure there is a great module page for it, and fix the corresponding piece of PyDrake to refer to it? Then repeat as needed?

EricCousineau-TRI, 2020-02-03:
Ah, by all means, we can certainly do it incrementally. I'm not vying for us to overhaul docs immediately here and now, just pointing out the immediate pain point for me.

My current plan for Python cross-referencing is to just make whatever reference is appropriate for the given module with zero modification to C++ docs; e.g. Solvers has a great overview, so I'd refer to it's Doxygen Module page; for Multibody, it has great module pages, but is a bit patchy in terms of connecting to API, so I'd just refer to the namespace directly.

On topic of organizing C++ docs in general, I will make a separate C++ issue. Regarding a *real* plan of attack, I'd prefer sketching that out in a Google Doc, but will make mention of that there.