RobotLocomotion/drake #12660: bazel run in tutorials errors out

RussTedrake, 2020-02-02:
Testing #12646 locally on my ubuntu machine demonstrated a failure to launch jupyter:
```
russt@Puget-179850-01:~/drake/tutorials$ bazel run rendering_multibody_plant
Starting local Bazel server and connecting to it...
INFO: Invocation ID: 4d6b6aea-e675-4fa1-b8e4-4cde0f9435dd
INFO: Analyzed target //tutorials:rendering_multibody_plant (197 packages loaded, 22541 targets configured).
INFO: Found 1 target...
INFO: Deleting stale sandbox base /home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/sandbox
Target //tutorials:rendering_multibody_plant up-to-date:
  bazel-bin/tutorials/rendering_multibody_plant_jupyter_py_main.py
  bazel-bin/tutorials/rendering_multibody_plant
INFO: Elapsed time: 253.006s, Critical Path: 184.69s
INFO: 428 processes: 428 linux-sandbox.
INFO: Build completed successfully, 449 total actions
INFO: Build completed successfully, 449 total actions
Running notebook interactively
[I 07:09:00.637 NotebookApp] Serving notebooks from local directory: /home/russt/.cache/bazel/_bazel_russt/a0bdb2099cb05916281dea472bfce61b/execroot/drake/bazel-out/k8-opt/bin/tutorials/rendering_multibody_plant.runfiles/drake/tutorials
[I 07:09:00.638 NotebookApp] The Jupyter Notebook is running at:
[I 07:09:00.638 NotebookApp] http://localhost:8888/?token=09ba3881df211e8e63672b610c39c30bd16750bae3e9422c
[I 07:09:00.638 NotebookApp] Use Control-C to stop this server and shut down all kernels (twice to skip confirmation).
Traceback (most recent call last):
  File "/usr/local/bin/jupyter-notebook", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/jupyter_core/application.py", line 266, in launch_instance
    return super(JupyterApp, cls).launch_instance(argv=argv, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/traitlets/config/application.py", line 658, in launch_instance
    app.start()
  File "/usr/local/lib/python2.7/dist-packages/notebook/notebookapp.py", line 1784, in start
    self.launch_browser()
  File "/usr/local/lib/python2.7/dist-packages/notebook/notebookapp.py", line 1742, in launch_browser
    with open(fd, 'w', encoding='utf-8') as fh:
TypeError: coercing to Unicode: need string or buffer, int found
```
The same error occurs when I try any of the other examples.
```
russt@Puget-179850-01:~/drake/tutorials$ jupyter notebook --version
5.7.4
```
I've also rerun `install_prereqs` and confirmed that this does not fix anything.

RussTedrake, 2020-02-02:
Update, it turns out that running `jupyter notebook [filename.ipynb]` in the tutorials directory also fails.  Running just `jupyter notebook` and then opening the files directly succeeds.
```
russt@Puget-179850-01:~/drake/tutorials$ jupyter notebook linear_program.ipynb 
[I 07:29:44.518 NotebookApp] The port 8888 is already in use, trying another port.
[I 07:29:44.523 NotebookApp] Serving notebooks from local directory: /home/russt/drake/tutorials
[I 07:29:44.523 NotebookApp] The Jupyter Notebook is running at:
[I 07:29:44.523 NotebookApp] http://localhost:8889/?token=7c4a30ecd10aafe6dfe0311ddb46e573bcb604a04eba7a4b
[I 07:29:44.523 NotebookApp] Use Control-C to stop this server and shut down all kernels (twice to skip confirmation).
Traceback (most recent call last):
  File "/usr/local/bin/jupyter-notebook", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/jupyter_core/application.py", line 266, in launch_instance
    return super(JupyterApp, cls).launch_instance(argv=argv, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/traitlets/config/application.py", line 658, in launch_instance
    app.start()
  File "/usr/local/lib/python2.7/dist-packages/notebook/notebookapp.py", line 1784, in start
    self.launch_browser()
  File "/usr/local/lib/python2.7/dist-packages/notebook/notebookapp.py", line 1742, in launch_browser
    with open(fd, 'w', encoding='utf-8') as fh:
TypeError: coercing to Unicode: need string or buffer, int found
russt@Puget-179850-01:~/drake/tutorials$ jupyter notebook
[I 07:30:26.465 NotebookApp] The port 8888 is already in use, trying another port.
[I 07:30:26.471 NotebookApp] Serving notebooks from local directory: /home/russt/drake/tutorials
[I 07:30:26.471 NotebookApp] The Jupyter Notebook is running at:
[I 07:30:26.471 NotebookApp] http://localhost:8889/?token=e51d1e61d2817796f4a95fd6667f3bf6611d9571348c6d11
[I 07:30:26.471 NotebookApp] Use Control-C to stop this server and shut down all kernels (twice to skip confirmation).
[C 07:30:26.475 NotebookApp] 
    
    To access the notebook, open this file in a browser:
        file:///run/user/1001/jupyter/nbserver-21900-open.html
    Or copy and paste one of these URLs:
        http://localhost:8889/?token=e51d1e61d2817796f4a95fd6667f3bf6611d9571348c6d11
Opening in existing browser session.
```

EricCousineau-TRI, 2020-02-03:
To me, this part of the error message:
```
Traceback (most recent call last):
  File "/usr/local/bin/jupyter-notebook", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/jupyter_core/application.py", line 266, in launch_instance
    return super(JupyterApp, cls).launch_instance(argv=argv, **kwargs)
```

Indicates that (a) this is Python 2.7 and (b) it is possibly pulling in a version of Jupyter from PIP.

Can you try to uninstall Jupyter in Python 2 (not sure if it's Homebrew or PIP), and then try to install Jupyter only in Python 3?

@RussTedrake If you run Drake's `install_prereqs`, is it successful?

@jamiesnape Is this something that our current `install_prereqs` for Mac could be missing?

jamiesnape, 2020-02-03:
> @jamiesnape Is this something that our current install_prereqs for Mac could be missing?

Not really. It is a common problem for both Python 2 and Python 3 versions of packages to both write to the same file in `bin`. We can't stop people from accidentally overwriting files after they have run `install_prereqs`.

EricCousineau-TRI, 2020-02-03:
Makes sense.

@RussTedrake In this case, I'd recommend you purge `jupyter-notebook` / `jupyter` from all package managers (Homebrew, PIP), then just have it installed via `install_prereqs`.

RussTedrake, 2020-02-06:
That did not resolve it.  On ubuntu I did 
```
sudo pip uninstall jupyter            
```
which did some removing, and then confirmed that every variant (pip2, jupyter-notebook) now all say "not installed".  I then ran `install_prereqs.sh` and see the same behavior.

jamiesnape, 2020-02-06:
@RussTedrake Do your underactuated notebooks work correctly through Bazel on that machine?

jwnimmer-tri, 2020-02-06:
FYI On Ubuntu, this is what a pipless machine should look like:
```console
jwnimmer@cons:~$ find ~/.local/lib
find: ‘/home/jwnimmer/.local/lib’: No such file or directory
jwnimmer@cons:~$ find /usr/local/lib
/usr/local/lib
/usr/local/lib/python3.6
/usr/local/lib/python3.6/dist-packages
/usr/local/lib/python2.7
/usr/local/lib/python2.7/dist-packages
/usr/local/lib/python2.7/site-packages
```

There are _no_ files.  There can be a few empty directories.

EricCousineau-TRI, 2020-02-06:
@RussTedrake When you have a chance, could you show me this error on your machine? I'm sure I could fix it in about 5min if I have direct access.

EricCousineau-TRI, 2020-04-02:
Just to check, is this still an issue?

RussTedrake, 2020-04-02:
I was able to run it on that same machine today, and everything worked.  Will close it now, and reopen if it ever rears it's head again.

closed 2020-04-02 by RussTedrake