RobotLocomotion/drake #12661: Add tutorials for different types of optimization

hongkai-dai, 2020-02-02:
I think we can add tutorials for the following type of optimization programs (one tutorial for each)

- [ ]  Linear Programming
- [ ] Quadratic Programming
- [ ] Second-order cone Programming
- [ ] Semidefinite Programming
- [ ] Sum-of-squares Programming
- [ ] Nonlinear Programming
- [ ] Mixed-integer Programming

And separate tutorials for
- [ ] inverse kinematics.
- [ ] Trajectory optimization.
- [ ] Lyapunov verification.

The goal is to complete the tutorials (LP, QP, MIP, IK, etc) before 6.832 students need them (probably before they submit the 6.832 project proposals?) I think the tutorials on trajectory optimization and Lyapunov are less urgent, and 6.832 already have examples on these.

@RussTedrake what do you think?

RussTedrake, 2020-02-03:
Wow.  that's a lot, but it's incredible.  I do think there is value in adding each of those MP tutorials, but I would actually rate the kinematics / inverse kinematics and trajectory optimization as the most important.  I do have a fair number of lyapunov examples (+ my ROA code will come in soon), but no kinematics examples, and it would be nice to lay out the types of trajectory optimization we already support in drake.

hongkai-dai, 2020-02-03:
Got it, I will add the IK and trajectory optimization tutorial first.

RussTedrake, 2020-02-03:
I think even an MBP kinematics + dynamics (including showing off symbolic, like we do in 
https://mybinder.org/v2/gh/RussTedrake/underactuated/master?filepath=examples%2Fdouble_pendulum%2Fdynamics.ipynb ) would be awesome.  we could even port the double pendulum example...