RobotLocomotion/drake #12674: Context::SetTimeStateAndParametersFrom() should copy fixed input port values

sherm1, 2020-02-04:
When the source context has fixed input ports, their values should be copied along with the rest of the source context's contents. Diagram contexts should find all the internal subsystem fixed inputs and copy them too. Note that unlike connected input ports, fixed ports can be copied with no System available for evaluating them (and this method does not have access to a System).

Note that there is already a System::FixInputPortsFrom() that is available to _evaluate_ all input ports to obtain values for them, whether fixed or connected in the source.

I'm resurrecting this from a long-ago discussion in issue #5872. Everything is much clearer now than it was then!

sherm1, 2020-02-04:
Consider renaming this method to something like `CopyAllValueFrom(source_context)`.

sherm1, 2020-06-02:
Related: #11351

sherm1, 2020-06-02:
Q: A connected (not fixed) input port whose value is currently up to date could also become a fixed input port in the destination without requiring System help (that is, with just the source Context). Should it be?

SeanCurtis-TRI, 2020-06-02:
This presumes copying a sub-context of a larger diagram? It seems to me that it should be parameter-controlled. It's easy to lose track of what context you have and what context it exists. So, someone cloning a sub-context by accident (mistakenly believing that they've done the right thing), but "succeeding"  in some sense seems unfortunate. But a user who knows they are cloning a sub-system and severing ties should be able to declare that and get that outcome.

sherm1, 2020-06-02:
The most common use I've seen (my visibility is low on this though) is to copy a whole Context after transmogrification. (The method copies a `Context<U>` into a `Context<T>`.)

xuchenhan-tri, 2020-09-18:
I'm going to take a stab at this issue. I'm currently learning more about the background, and any pointer/suggestion/opinion on how to proceed will be appreciated.

xuchenhan-tri, 2020-09-28:
There are two things that prevent Context to copy fixed input port values in SetTimeStateAndParametersFrom() without the help of the corresponding System:

(1) The Context is ignorant of the PortDataType of the input port, so it doesn't know whether to copy the fixed input port values over as numerical values or abstract values.
(2) If the fixed input port value being copied over is abstract and scalar-dependent, there is no way to transmogrify the underlying data to the correct scalar type.

(1) can be solved by passing the PortDataType information to the Context at the time the input port is constructed.
(2) can be solved when [#5454 ](https://github.com/RobotLocomotion/drake/issues/5454) is resolved.