RobotLocomotion/drake #12678: Update all macOS CI builders to bazel 2.0

jwnimmer-tri, 2020-02-04:
Some builders are reporting bazel 1.2.1 currently.

jwnimmer-tri, 2020-02-04:
1.2.1:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-catalina-clang-bazel-nightly-debug/111/

2.0.0:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-mojave-clang-bazel-nightly-release/203/

Looks like catalina vs mojave, perhaps.

jamiesnape, 2020-02-04:
That is very strange. Looking into it now.

jwnimmer-tri, 2020-02-04:
FWIW If new images are coming, might be worth PRing this as well:
https://github.com/RobotLocomotion/drake/blob/2bfc34f1d24cfbc168ea8d0403c81fb3ce559e69/setup/mac/binary_distribution/install_prereqs.sh#L22-L24

jamiesnape, 2020-02-04:
The images were there, but for some reason only 3 of 4 were deployed. User error on my part, it seems.

jwnimmer-tri, 2020-02-06:
Fixed now.

closed 2020-02-06 by jwnimmer-tri