RobotLocomotion/drake #12697: Adding a simple ForceElement to an MBP is very hard

ggould-tri, 2020-02-06:
(This is a metabug; subsidiary issues should be opened for its contributing factors if they are judged sufficiently individually buggy)

A real-world system I am modeling has a hinge with complex dynamics (static, dynamic, and viscous friction, a catch, and a spring).  I decided to model this by creating a `ForceElement`, and wrote the necessary four functions (force, energy, non-conservative power, and conservative power) on `double` as well as a class that took a Joint to apply this to.

The resulting odyssey is chronicled on slack:
 * https://app.slack.com/client/T0JNB2DS4/C2WBPQDB7/thread/C2WBPQDB7-1581012046.027400
 * https://app.slack.com/client/T0JNB2DS4/C2WBPQDB7/thread/C2WBPQDB7-1581012046.027400
 * https://app.slack.com/client/T0JNB2DS4/C2WBPQDB7/thread/C2WBPQDB7-1581012046.027400

The major problems that I encountered:
 * `MultibodyPlant::AddForceElement` will not accept a simple concrete object but rather demands fowarded arguments to a class template ctor (ie, you have to include all of the scalar clone boilerplate).
 * `ThrowIfAlgebraicLoopExists` will not accept a template class that throws on clone (ie, you actually have to *implement* all of the scalartypes). (see also #12253)
 * Taking a `Joint` in the ctor makes the class un-cloneable, as you cannot resolve a `joint_id` at clone time (the joints in the cloned MBT having not yet been constructed).
   * However you also can't invariant-check or even typecheck that `joint_id` when you receive it -- the obvious "dereference the ID and dynamic_cast it" check asserts.
 * This was of course augmented by the perpetual defect that our doxygen pages show protected methods you can't call but not the superclass methods that you must call instead.
 * Finally, per @jwnimmer-tri, "the power and energy numbers go nowhere, do nothing.  You can return NAN for them if you want" -- ie half of the two dozen non-boilerplate lines of PR were also unnecessary, but for an undocumented reason.

Of course each of the above also suffers from its own incomprehensibility problem:
 * Dereferencing a joint too early gives an out-of-range assertion that you have to read `multibody_tree.cc` to understand.
 * Error messages for forwarded args are so bad that IMO even using the idiom should probably be defective.
 * Doxygen is not a useful guide to any of this.

Which of these problems are or aren't worth solving -- and if so how -- will doubtless be a matter of some controversy.  However I think there is a meta defect here that we have made a straightforward matter (a couple of dozen lines of math) into a journey through the heart of darkness of forwarding constructors, templates of templates, and the objects-versus-IDs mess.  An object which could conceptually be two symbolic expressions (energy, nonconservative power) and a correspondence to joint {q,v} (or even just one function of double, given that energy and power are unused) is instead somewhat north of our maximum code-review length even without tests.

ggould-tri, 2020-02-06:
(I will attach code here once it's stable -- although this is from a TRI repo, it is written in pure Drake with no TRI dependencies)

jwnimmer-tri, 2020-02-13:
Another complaint... The entire `ForceElement` API towards its subclasses mentions `multibody::internal` namespace frequently -- for the `MultibodyTree`, `PositionKinematicsCache`, etc.

Since users shall not touch things in `internal` namespace, it's not possible to legally implement a new `ForceElement`.

jwnimmer-tri, 2024-11-01:
See #13350 for some additional background.