RobotLocomotion/drake #12714: pyplot_animation_multibody_plant tutorial fails on binder

RussTedrake, 2020-02-11:
https://notebooks.gesis.org/binder/jupyter/user/robotlocomotion-drake-t607tgop/notebooks/tutorials/pyplot_animation_multibody_plant.ipynb

To reproduce: execute down to the cell "Run without playback".  You will see:

```
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-6-7ddd7ff471a2> in <module>()
----> 1 run_pendulum_example(playback=False)

<ipython-input-5-28fecc64cfd9> in run_pendulum_example(duration, playback, show)
     21     visualizer = builder.AddSystem(PlanarSceneGraphVisualizer(
     22         scene_graph, T_VW=T_VW,
---> 23         xlim=[-1.2, 1.2], ylim=[-1.2, 1.2], show=show))
     24     builder.Connect(pose_bundle_output_port, visualizer.get_input_port(0))
     25     if playback:

TypeError: __init__() got an unexpected keyword argument 'show'
```

I suspect this means that binder is using a stale version of drake (as we added the show argument fairly recently, but > 1 week ago now).

EricCousineau-TRI, 2020-02-11:
I was able to run this tutorial without a hitch...
Can you try a fresh user session? (possibly using a Private Browsing session to assure there's no connection via cookies or what not?)

jamiesnape, 2020-02-11:
The reason here is that `underactuated` did not change after the `drake` image was updated, so Binder did not rebuild its image. Not using `latest` and bumping the `drake` docker tag when a particular feature is needed by `underactuated` is the solution here.

jamiesnape, 2020-02-11:
(Or you ensure a commit in `underactuated` happens after a new feature in `drake` is required. But if you are going to do that, then you may as well make the commit in question update the docker tag.)

RussTedrake, 2020-02-12:
I don't think this has anything to do with underactuated?  I launched it directly from the drake.mit.edu start page?  And it is still happening now on a different machine, etc.

EricCousineau-TRI, 2020-02-13:
Can I ask which browser / OS you've tried, and if you've tried a private browsing session? (to ensure Binder gives you a new session?)

jamiesnape, 2020-02-14:
> I don't think this has anything to do with underactuated? I launched it directly from the drake.mit.edu start page? And it is still happening now on a different machine, etc.

Ok, wrong issue. We had the same problem in underactuated, however.

EricCousineau-TRI, 2020-03-19:
Since we're unable to repro this, will close this for now. Russ, if you're able to recreate, please let us know, along with browser / OS + trying a fresh session.

closed 2020-03-19 by EricCousineau-TRI