RobotLocomotion/drake #12722: Allegro example broken with new default discrete_update_period

mattcorsaro1, 2020-02-11:
## Problem
In the allegro example `allegro_single_object_simulation`, [max_time_step](https://github.com/RobotLocomotion/drake/commit/cb63eb8bb8aeb7dbc7b623144423201b2b1d2efe#diff-4d8d1fbade285ceab0028add90f04599L48) was 1.5e-4 by default. [cb63eb8](https://github.com/RobotLocomotion/drake/commit/cb63eb8bb8aeb7dbc7b623144423201b2b1d2efe) replaced it with [mbp_discrete_update_period](https://github.com/RobotLocomotion/drake/commit/cb63eb8bb8aeb7dbc7b623144423201b2b1d2efe#diff-4d8d1fbade285ceab0028add90f04599R52), which is 1.0E-3 by default.

When the `run_twisting_mug` example is run to close the hand, `allegro_single_object_simulation` crashes with this error:
```
abort: Failure at multibody/plant/tamsi_solver.cc:217 in SolveQuadraticForTheSmallestPositiveRoot(): condition 'Delta > 0' failed.
```

## Steps to reproduce
### failure
`bazel-bin/examples/allegro_hand/joint_control/allegro_single_object_simulation`
`bazel-bin/examples/allegro_hand/joint_control/run_twisting_mug`

### success
`bazel-bin/examples/allegro_hand/joint_control/allegro_single_object_simulation --mbp_discrete_update_period 1.5e-4`
`bazel-bin/examples/allegro_hand/joint_control/run_twisting_mug`

(This was the only instance of the default value being changed that I noticed in this commit.)

## Fix
Change the default value of mbp_discrete_update_period back to 1.5e-4. Implemented [here](https://github.com/mattcorsaro1/drake/commit/ee86570c57defd365efea4b37b2ac2de794e350a) in [my branch](https://github.com/mattcorsaro1/drake/tree/allegro_update_period).

amcastro-tri, 2020-02-12:
oh, good catch @mattcorsaro1. Your fix is correct. You ok with pushing it to master? I can do it also if you want.

mattcorsaro1, 2020-02-12:
If you could do it, that'd be great. Thanks!

jwnimmer-tri, 2020-02-12:
Remember that the PR will also need a test (perhaps just an `add_test_rule = True`).

jwnimmer-tri, 2020-02-13:
I'll do this one; it turns out that the regression test will be a bit novel.

PR #12734 (2020-02-14): allegro_hand: Use correct timestep and add regression test
Closes #12722.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12734)
closed 2020-02-14 by jwnimmer-tri