RobotLocomotion/drake #12747: MacOS docutils version mismatch

mposa, 2020-02-18:
A few students here at Penn have run into this issue on MacOS Catalina. During the prereq installation, pip3 installation of `botocore` fails because it requires `docutils < .16`, whereas these students have all had version .16 installed. Manually downgrading docutils resolved the issue.

jamiesnape, 2020-02-18:
Our scripts will install whatever `botocore` needs (we don't have a dependency on `docutils`). We cannot cover every potential pre-existing configuration, unfortunately.

jamiesnape, 2020-02-18:
Relates https://github.com/boto/botocore/issues/900 https://github.com/boto/botocore/issues/1789, https://github.com/boto/botocore/issues/1796, and maybe a few others.

mposa, 2020-02-18:
From my experience this week, this extends beyond "every potential pre-existing configuration," as 4 of 4 Mac users this week have this issue.

jamiesnape, 2020-02-18:
On a given day it will be different. We cannot legislate for `botocore`. You can complain about the pin to them. If we pin artificially, then we shift the problem too.

jamiesnape, 2020-02-18:
Also I need to know the common package (if any) that is installing docutils that they all have to even have a chance of lessening the impact. Try `pipdeptree` or similar.

blandry, 2020-02-18:
For what it's worth I also ran into this. Like Michael got around it easily but suspect this will unfortunately come up a few time among your users.

prometheanfire, 2020-02-21:
this is not a mac issue, it's a they should only cap when future versions are KNOWN to break only.  Having this happen every few weeks (to this and other libs) is getting old.

pkg_resources.ContextualVersionConflict: (docutils 0.16 (/home/zuul/src/opendev.org/openstack/requirements/.tox/py36-check-uc/lib/python3.6/site-packages), Requirement.parse('docutils<0.16,>=0.10'), {'botocore'})

prometheanfire, 2020-02-21:
See https://github.com/boto/botocore/issues/1872#issuecomment-550500977 as well, I think this is fairly universally looked down upon as a practice.

prometheanfire, 2020-02-21:
if you need to cap docs, use a requirements file for docs jobs that's separate from a normal install.  See https://github.com/openstack/nova/blob/master/doc/requirements.txt for example

jwnimmer-tri, 2020-02-24:
I agree with @jamiesnape that the right answer is to complain to botocore maintainers, and there is not much (if anything) we can do to work around the problem.

The only worthwhile change might be to split drake's dependencies from only "binary" vs "source" to "binary" vs "user source" vs "developer source" where any dependencies not used by tests in the default configuration are deferred to "developer source".  That would be things like clang, valgrind, boto, etc.  That would require add complexity CI, but would _substantially_ reduce the build-from-source dependency footprint.

It would also happen to sidestep the botocore problem, because the default test suite does not use boto for anything AFAICT.

RussTedrake, 2020-03-07:
fwiw -- today i saw this error scroll by during a fresh install on a clean mac machine, but it did not prevent `install_prereqs` from completing.   I was able to `bazel test //...` and also `install`, and saw no problems.

jwnimmer-tri, 2020-05-15:
> ... The only worthwhile change might be to split drake's dependencies from only "binary" vs "source" to "binary" vs "user source" vs "developer source" where any dependencies not used by tests in the default configuration are deferred to "developer source" ...

That's basically what #13168 is proposing for other reasons, so I'll resolve this as a duplicate of that.

closed 2020-05-15 by jwnimmer-tri

jwnimmer-tri, 2020-12-17:
Also, as of https://github.com/boto/botocore/issues/900 (botocore 1.18), the docutils dependency is gone.

jamiesnape, 2020-12-17:
Not this precise issue, but they still have their bad habit of pinning dependencies fairly tightly. Still on my troublesome list of modules.