RobotLocomotion/drake #12749: install: Should support `//:drake_shared_library` for bazel external workflow

EricCousineau-TRI, 2020-02-18:
During the Drake Developers meeting, @mposa mentioned that compilation time is a pain point for using Drake in a Bazel-centric workflow.

Given that we now have `@drake//bindings/pydrake` exposed ([example here](https://github.com/RobotLocomotion/drake-external-examples/tree/0dbb55513454fa4011f14907608433d03e4796b9/drake_bazel_installed)), we should also expose `@drake//:drake_shared_library` to allow C++ compilation.

Some possible challenges:
- Ensuring that third-party libraries have their dependencies correctly enumerated
- There may not be 1-to-1 parity between the from-source and from-binary `drake_shared_library`, so there may be some new failure modes.

As a side note, once #7294 / #6760 are done, we should be able to expose those more granular targets in the from-binary Bazel consumption of Drake.

\cc @jwnimmer-tri

jamiesnape, 2020-02-18:
Indeed. There is a lot we do not expose. Probably not going to happen before Bazel switches to `@rules_cc` (though that should be soon).

EricCousineau-TRI, 2020-05-23:
@mposa From looking at y'all's code in [dairlib](https://github.com/DAIRLab/dairlib/tree/5520e9f5c6345d161eed0c81ce4707e1d459ca9b), would y'all be a potential consumer of this, were it to happen?

(It would hurt target granularity, esp. if you have your own Drake version; however, it would kinda simplify stuff like ODR stuff with Python bindings.)

mposa, 2020-05-23:
If I understand, then yes we would be happy consumers. To get pybind working with our code, we had to throw out Drake dependency granularity (everything now depends on all of Drake), so if we could avoid having to compile Drake I'm guessing it would massively improve our compile times (except in the rare cases where someone's tinkering with Drake).

guzhaoyuan, 2020-08-24:
I find this issue relevant to my case as well. 

I am looking for ways to import drake as an external binary and build c++ with bazel workflow. I try to use binary because my Mac is not a powerful machine and it takes forever to compile drake.

I looked at the [drake-external-examples here](https://github.com/RobotLocomotion/drake-external-examples/tree/master/drake_bazel_installed) but could not find a routine to compile c++ using drake binary with bazel.

One more petition for the `drake_shared_library` here!

PR #16072 (2021-11-15): [install] Add :drake_shared_library to drake-bazel-installed
Remove deprecated py_repositories() from unit test.

Closes #12749.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16072)
closed 2021-11-15 by jwnimmer-tri