RobotLocomotion/drake #12754: Should be possible to provide sdf/urdf models as strings

ggould-tri, 2020-02-20:
Currently the only way to add SDF/URDF models to a MultibodyPlant is via `AddModelFromFile`.  However there are cases where a model is more easily provided in the form of a string or a stream.  For example @avalenzu notes that a DRC team had the robot provide its model over the wire, and I am currently working with a robot (WLKATA Mirobot) that emits the link data used in its internal IK via its serial protocol.

It would probably make sense to provide AddModelFromStream and a trivial wrapper around that for AddModelFromString.

jwnimmer-tri, 2020-08-04:
The libsdformat APIs don't really support streaming, so I just plan to add a string overload directly.

SeanCurtis-TRI, 2020-08-04:
@jwnimmer-tri I have a branch where I started this as well. Are you actively tackling this? I note you elevated the priority. Depending on how far along you are, I can either finish mine or retire it.

SeanCurtis-TRI, 2020-08-04:
Never mind; I just noticed the PR.

EricCousineau-TRI, 2020-08-04:
FWIW Semi-relates #9500 (stuff on the wire, etc)

jwnimmer-tri, 2020-08-05:
I added `AddModelFromString` but not the plural `AddAllModelsFromString` companion.  I'll reopen the issue (at lower priority) to go back and add that, too.

rpoyner-tri, 2022-03-18:
Is there a reason this issue is still open? the String APIs have been available for quite some time.

PR #13814 (2020-08-05): parsing: Add support for parsing SDF and URDF from strings
Closes #12754.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13814)
closed 2022-03-18 by rpoyner-tri