RobotLocomotion/drake #12757: Velocities start index for floating bodies should index velocities, not state.

amcastro-tri, 2020-02-21:
Currently `Body::floating_velocities_start()` returns an index into the full multibody state vector.
Contrast this API to `Joint::velocity_start()` which does return an index into the vector of generalized velocities. This only confuses users.

@samzapo proposes that  `Body::floating_velocities_start()` behaves exactly as `Joint::velocity_start()`, returning an index into the vector of generalized velocites.

I'd like to hear feedback from other users of this API. Possibly @ggould-tri, @hongkai-dai, @avalenzu, @mposa 

cc'ing @edrumwri

edrumwri, 2020-02-21:
I'm in favor of this proposal.

edrumwri, 2020-03-26:
Can I upvote work on this? Just got bitten by this, even though I knew about it!

amcastro-tri, 2020-03-26:
Relates to #12807

samzapo, 2022-02-25:
@amcastro-tri I just stumbled upon this issue again.
`Body`
```
  /// (Advanced) For floating bodies (see is_floating()) this method returns the
  /// index of the first generalized velocity in the state vector for a
  /// MultibodyPlant model.
  /// Velocities for this body are then contiguous starting at this index.
  /// @throws std::exception if called pre-finalize, see
  /// MultibodyPlant::Finalize().
  int floating_velocities_start() const {
    DRAKE_BODY_THROW_IF_NOT_FINALIZED();
    return topology_.floating_velocities_start;
  }
```
and then `Joint`
```
  /// Returns the index to the first generalized velocity for this joint
  /// within the vector v of generalized velocities for the full multibody
  /// system.
  int velocity_start() const {
    return do_get_velocity_start();
  }
```

amcastro-tri, 2022-02-28:
ok, lets fix this thing.
My only concern here is how to go about deprecation since clearly changing the behavior of this method would entail a breaking change.
Any recommendations @sherm1, @jwnimmer-tri  on how to go about this?
That is, if we want (and we do) to change the behavior of `Body::floating_velocities_start()` (to return a completely different index), what is the process to land such a breaking change?

jwnimmer-tri, 2022-02-28:
I see.  The path forward is to add a new function name with the semantics you want, and deprecate the function with the old name.

amcastro-tri, 2022-02-28:
The problem is that I think we'd like to keep the "old name". 
Unless @samzapo or @sherm1  have other name suggestions in mind?

jwnimmer-tri, 2022-02-28:
Using a slightly more awkward name is a low price to pay in order to avoid breaking your users.

sherm1, 2022-02-28:
I agree we should go with an awkward name first. For example we could do this:
- add `floating_velocities_start_in_v()` with the correct behavior
- deprecate `floating_velocities_start()`
- later, remove `floating_velocities_start()`

Then if we hate the new name:
- add `floating_velocities_start()` with the correct behavior, deprecate `floating_velocities_in_v()`
- later, remove `floating_velocities_in_v()`

drewhamiltonasdf, 2023-10-29:
Liking this conversation about making indexing into large plants easier. I just wanted to point out to @sherm1 who seems to be fixing a bunch of things in Multibody plant right now another gripe, which is that velocity_start() and position_start() return an index of 0 for "weld" joints. In my view they should return None as there is no valid index into the multibody state vector for a zero-dof joint. The fact that these typically come at the end of of the JointIndices aggravates things as the weld joints will consistenly overwrite other valid states if you aren't paying attention. Hope that all makes sense.

jwnimmer-tri, 2023-10-29:
See #20428.

sherm1, 2023-10-30:
Two items here:
- For the original topic of this issue, I'll fix floating_velocities by adding _in_v() and _in_state() and deprecating the confusing name. After that's safely removed we can reintroduce it with the correct behavior.
- Regarding weld numbering: #20428 changed numbering to be consistent with other joints. That means the start coordinates are reported as being where in the state the coordinates would have started if there were any (based on structure of the multibody tree). "None" might make more sense in Python, though it would require the caller to special-case for welds. Using the fact that num_positions() and num_velocities() return 0 for welds usually allows all the joints to be treated uniformly without special casing.

PR #20454 (2023-11-01): [multibody] Deprecate confusing Body::floating_velocities_start() method
In MbP Joint and in MbP/MbT internals, `velocity_start()` means "start index in v" while in MbP Body, `floating_velocities_start()` means "start index in state qv" so adds num_positions() to skip over q. This has been an endless source of confusion, as first noted by @amcastro-tri in #12757. The "within state" indexing also bakes in an assumption that we store q and v concatenated, something we may want to change for performance reasons later.

This PR deprecates  `Body::floating_velocities_start()` and replaces it with ` [... cut]
closed 2023-11-01 by sherm1