RobotLocomotion/drake #12759: Port the DoorHinge functionality from Anzu to Drake

huihuaTRI, 2020-02-21:
@ggould-tri has recently implemented a force element `dish::DoorHinge` in Anzu to simulate the hinged door of a dishwasher. This is a very useful feature and could be used for other assets with doors. It has already been used in a different simulation since then.

We would like to add this feature to Drake. Additionally, a stop joint could potentially be added to the DoorHinge by adding another curve to simulate the stopping force when the joint hits the stop limit. I just want to throw this idea. The stop joint does not necessarily need to be implemented in the same PR.

ggould-tri, 2020-02-24:
Thanks for creating this issue!  For reference a recent version of the code is at https://gist.github.com/ggould-tri/b686cf978754bc2bffe55ee99daff8a5 .

I've been thinking a bit about how to generalize this functionality and I think that a generalized mechanism for ForceElement(Joint, Symbolic expression of conservative energy, Symbolic expression of nonconservative power) would be a clear and concise factoring for Drake; the door hinge could then be a special case (or example) of this.  The force expressions could be obtained by symbolic differentiation.

I know that Russ also has concerns about my use of tanh rather than stribeck-like piecewise functions that should probably at least get an airing.

I'll attend (at least part of) this afternoon's dynamics meeting; we can perhaps give this some more thought then.

sherm1, 2020-02-24:
> I know that Russ also has concerns about my use of tanh rather than stribeck-like piecewise functions that should probably at least get an airing.

Agreed we should discuss. I think `tanh()` is the better choice.

ggould-tri, 2020-02-24:
(fyi @RussTedrake for reference ^^^ )

RussTedrake, 2020-02-27:
Just to be clear... I haven't looked closely enough at this to have an opinion on it which form.  But I do think that we should offer joint static friction -- in some form -- for any MBP.

sherm1, 2020-03-13:
@ggould-tri out of curiosity, what led you to choose ForceElement for the DoorHinge rather than making a separate Drake LeafSystem? I'm happy with the ForceElement approach but wonder what are the tradeoffs between those approaches.

ggould-tri, 2020-03-13:
I looked at the existing APIs for applying a force to an MBP, I saw ForceElement, and I concluded that was the API to apply a force, so I used it.  Figuring out how to wire up a raw LeafSystem to a particular joint didn't even occur to me.

sherm1, 2020-04-02:
Done in PR #12812 -- thanks, @huihuaTRI !

closed 2020-04-02 by sherm1