RobotLocomotion/drake #12767: [geometry] QueryObject introduces Eval...() methods (cache query results)

SeanCurtis-TRI, 2020-02-25:
# Problem Statement

`QueryObject` is the main interface through which other entities can make geometric queries. Invoking a query method on that object always leads to calculations. Therefore, if downstream entities have multiple code paths that would depend on the same query, and *care* about performance, they are obliged to cache the results of the query. This is unfortunate. It doesn't scale well and displaces `QueryObject`'s rightful burden.

# Proposed solution

`QueryObject` should provide `Eval()` versions of those queries (or as many as is reasonable) such that downstream callers are free to evaluate the same query over and over and not worry about incurring a cost. Very much in line with the current `Calc()` vs `Eval()` paradigm.

SeanCurtis-TRI, 2020-02-25:
@amcastro-tri FYI

sherm1, 2020-02-25:
Seems like a great idea!

jwnimmer-tri, 2024-03-20:
Closing until we have a motivating benchmark.

closed 2024-03-20 by jwnimmer-tri