RobotLocomotion/drake #12769: MultibodyPlant's CalcReactionForces() method should be public

sherm1, 2020-02-25:
Currently the only way to get joint reaction forces from MBP is to evaluate the reaction force output port. That port's cache entry uses `MBP::CalcReactionForces()`.

That API (or something like it) should be available for users to call. @ggould-tri has a use case for getting reaction forces in an event handler (for sample-and-hold). Currently the only way to do that is to evaluate the reaction_forces_output_port, which is obscure, and for which it is hard to find API-style documentation.

One nice thing about the output port is that it is cached. I think it would be fine to implement `MBP::EvalReactionForces(context)` internally by evaluating the output port's cache entry. That would make it free if the output port had been recently evaluated.

amcastro-tri, 2020-02-26:
IMO the API is there @sherm1, it's the output port you mention. Maybe a doc improvement to make it easier to discover? Maybe @ggould-tri can provide more details on how that'd help his problem.

ggould-tri, 2020-02-26:
My situation is that I am attempting to discern which of three awful hacks to get a joint's normal force is least awful: (1) Add a force gauge spring (#12775); (2) Wait for bushing support in drake-sdf (#11946 plus whatever future PR gives it SDF support); (3) Copy reaction forces out of the prior timestep's context (this issue).  So this issue would be stated as:

"There should be a clear and documented way to extract the reaction forces for a joint from a context for the purpose of storing them in state for reuse in a force element in the next timestep."

My current thinking is that this is a too horrible a hack to use if I can get the force gauge working, but this is my second alternative.

closed 2024-11-02 by jwnimmer-tri