RobotLocomotion/drake #12772: Capability to update joint types after loading a model from URDF

huihuaTRI, 2020-02-26:
Under certain circumstances, the users may need to change a joint type (for revolute to fixed for example) after loading a model from URDF files. The following are two use cases.
1. In the URDF files, the users may define multiple unactuated joints that could be used to adjust the position of certain parts. When actually running the simulation, these joints will be fixed to speed up the simulation.
2. A mobile base robot has an unactuated caster. After welding the robot to the ground, fixing this caster joint yields a fully actuated robot.  Then we could apply Drake inverse dynamics controller for this fully actuated robot.

There are multiple options for this feature request
1. Quote of Alejandro "Probably what we need is some sort of intermediate representation of the model. We could then parse a file into that representation. Then there'd be a free method that could take that model and perform all sorts of changes an simplifications. Then we'd build the actual plant from that."
2. Add the API directly to the current MBP structure.

sherm1, 2020-02-26:
Another option would be to have a "lock this joint" feature. That allows positioning via joints and kinematic analysis first, then locking allow treating those joints as prescribed (making for a very fast mixed forward/inverse dynamics calculation, and providing for free the joint forces needed to hold the joints locked).

Simbody has a generalized version of joint locking that is awesome (if I do say so myself :). Scroll down to "Mobilizer locking and unlocking" [here](https://simbody.github.io/simbody-3.6-doxygen/api/classSimTK_1_1MobilizedBody.html). Terminology translation: 
State-->Context, u/udot-->v/vdot.

amcastro-tri, 2020-02-27:
Well, that comment I made is a bit out of context. Consider what I said before: "I think we'd definitely like to have the capability to change the plant's topology. In your case however, it doesn't seem changing the topology of a complex plant is what we want."

But I also understood your comment "2) faster simulation" as, for instance, being able combine chains of weld joints into a single one (I've seen large models where for some reason this happens).

That being said, is this an optimization issue? because an SDF file does, for us at least now, constitute this "intermediate representation". See this [great snippet of code](https://github.com/RobotLocomotion/drake/compare/master...EricCousineau-TRI:issue-12775-wip) by @EricCousineau-TRI in which he switches weld by prismatic joints.

amcastro-tri, 2020-02-27:
The "lock this joint" method could work I believe. Would that solve your problem @huihuaTRI?

huihuaTRI, 2020-02-27:
In the link (https://github.com/RobotLocomotion/drake/compare/master...EricCousineau-TRI:issue-12775-wip) you sent, I don't see where he switches weld by prismatic joints. Maybe it's still in progress?

"lock the joint" would solve my problem. The current solution I took is the brutal force way: modify the urdf to fix those joints. All in all, what we discussed here would be a good feature to have.

EricCousineau-TRI, 2020-03-17:
FTR It effectively happens here, which is what you mentioned as the brute force way:
https://github.com/EricCousineau-TRI/drake/blob/7f375bb60540d3a51b2619bba474748fd186e52c/tmp/model_check.py#L144

But yeah, at some point it would be nice to be able to mutate (un-finalize / delete+add / re-finalize?) the plant which we've spoken about.

jwnimmer-tri, 2022-05-02:
Given the discussion above, I'll claim that joint locking (#14200) has sufficiently resolved this issue.

If we have a use case that _actually_ needs joint swapping post-parsing, we can open a new issue to explain it.

closed 2022-05-02 by jwnimmer-tri