RobotLocomotion/drake #12773: Getting Drake to work with ROS using find_package()

JeffR1992, 2020-02-26:
I've had a look at the official Drake website and on this Github too, but unfortunately haven't been able to find a direct answer to the following question (although there have been a few questions regarding `find_package()`). 

I downloaded the `drake` binaries from `https://drake.mit.edu/from_binary.html` and am trying to use some of `drake`'s functionality in a ROS project (in particular, `drake`'s implementation to efficiently solve the discrete algebraic Riccati equation (DARE)).

I'm on Ubuntu 16.04 using ROS Kinetic. As such, in my ROS project's `CMakeLists.txt` I have,
```
find_package(catkin REQUIRED COMPONENTS
  roscpp
  std_msgs
  mav_msgs
  drake
)
```
and in the `package.xml` I've added,
```
  <build_depend>drake</build_depend>
  <build_export_depend>drake</build_export_depend>
  <exec_depend>drake</exec_depend>
```
However, when calling `catkin build` in my workspace I get the following error,
```
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "drake" with any of
  the following names:

    drakeConfig.cmake
    drake-config.cmake

  Add the installation prefix of "drake" to CMAKE_PREFIX_PATH or set
  "drake_DIR" to a directory containing one of the above files.  If "drake"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:10 (find_package)
```
Is there a way to allow `find_package()` to work when trying to find the `drake` package?

jamiesnape, 2020-02-26:
> Ubuntu 16.04 using ROS Kinetic

Note that we have not supported 16.04 since 105877e57b253343d5591e6878ed135ad4a3330a (October 24, 2019). Assume you are using a version from before then, you are going to need to do exactly what the error message says (set `CMAKE_PREFIX_PATH` to `/path/to/drake` or set `drake_DIR` to `/path/to/drake/lib/cmake/drake`). Even then, you may run into issues since #9500 is unresolved.

JeffR1992, 2020-02-26:
Thanks @jamiesnape, I'm using the `drake` binaries from `drake-latest-xenial.tar.gz` (for some reason permission is denied when trying to download `drake-20191026-xenial.tar.gz`). I tried adding the explicit path to `drake` using `set(CMAKE_PREFIX_PATH "/home/jeff/dev/drake")` above the call to `find_package()` in `CMakeLists.txt`, but that still doesn't seem to work when calling `catkin build`.

I feel like I must be doing something incorrectly, the first few lines of my `CMakeLists.txt` now looks as follows,
```
cmake_minimum_required(VERSION 2.8.3)
project(simple_controller)

set(CMAKE_PREFIX_PATH "/home/jeff/dev/drake")

find_package(catkin REQUIRED COMPONENTS
  roscpp
  std_msgs
  mav_msgs
  drake
) 
```

jamiesnape, 2020-02-26:
You would need literally `find_package(drake REQUIRED)` and then use `target_link_libraries(.... drake::drake)`, etc. If that is not sufficient, let me know.

JeffR1992, 2020-02-26:
Hmm, `catkin build` complains about being unable to find `roscpp` with the above changes, which makes we think I did it incorrectly. This is my current `CMakeLists.txt` in its entirety:

```
cmake_minimum_required(VERSION 2.8.3)
project(quadrotor_controller)

set(CMAKE_PREFIX_PATH "/home/jeff/dev/drake")

find_package(catkin REQUIRED COMPONENTS
  roscpp
  std_msgs
  mav_msgs
)

find_package(drake REQUIRED)

catkin_package(
#  INCLUDE_DIRS include
#  LIBRARIES quadrotor_controller
  CATKIN_DEPENDS mav_msgs
#  DEPENDS system_lib
)

include_directories(
  include
  ${catkin_INCLUDE_DIRS}
)

add_library(simple_controller STATIC
  src/simple_controller.cpp
)

add_executable(${PROJECT_NAME}_node
  src/quadrotor_controller_node.cpp
)

target_link_libraries(${PROJECT_NAME}_node
  simple_controller
  drake::drake
  ${catkin_LIBRARIES}
)

set(CMAKE_CXX_COMPILER "/usr/bin/clang++")
```

jwnimmer-tri, 2020-04-28:
In the past couple of months, we've moved to using Stack Overflow as our desired forum for support (see https://drake.mit.edu/getting_help.html#asking-your-question).

I'm sorry for closing an issue without a clear resolution, but if you are still having trouble please open a new SO post tagged with drake, and provide the latest version of the failing example.

Note that current development is focused on Bionic and Focal; we haven't supported Xenial since last October at https://github.com/RobotLocomotion/drake/releases/tag/v0.11.0.

closed 2020-04-28 by jwnimmer-tri