RobotLocomotion/drake #12778: Missing an API to get JointActuatorIndex with a given joint name

huihuaTRI, 2020-02-27:
Say a user has a vector of PID gains for an under-actuated robot. The user may want to set different gains for different actuators. Given the actuator name, it would be convenient to have an API to be able to return the absolute index of that actuator. Something like 
`JointActuatorIndex actuator_index = GetJointActuatorByName(actuator_name).index()`.

Currently, the `JointActuator` class does not have such a member. Would `multibody_tree` support this feature easily? @amcastro-tri

amcastro-tri, 2020-02-29:
@huihuaTRI, I mislead you. We do have `MBP::GetJointActuatorByName()`.

closed 2020-02-29 by amcastro-tri

huihuaTRI, 2020-03-02:
I may not explain it clearly. We do have `MBP::GetJointActuatorByName()`, but we don't have anything to get the `JointActuatorIndex`.

jwnimmer-tri, 2020-03-02:
What about `GetJointActuatorByName(name).index()`?

huihuaTRI, 2020-03-02:
That returns a `JointIndex`, not a `JointActuatorIndex`. For an underactuated robot, these two will be different

jwnimmer-tri, 2020-03-02:
Really?  On paper at least it seems like `ElementIndexType = JointActuatorIndex`?  What am I missing?

```c++
template <template <typename> class ElementType,
    typename T, typename ElementIndexType>
class MultibodyElement {
 public:
  // ...
  /// Returns this element's unique index.
  ElementIndexType index() const { return index_;}

template <typename T>
class JointActuator final
    : public MultibodyElement<JointActuator, T, JointActuatorIndex> {
```

huihuaTRI, 2020-03-02:
I missed that one, Thanks for pointing it out. I will confirm whether it works and report it back here.

huihuaTRI, 2020-03-02:
Come back to confirm that `GetJointActuatorByName(name).index()` gives what I need. Thanks!

jwnimmer-tri, 2020-03-02:
So exactly as you request in the top post, at it happens!

Inheritance-oriented-programming strikes again.  It would be great if the https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_joint_actuator.html showed the base class methods.  We can probably add `using` statements in joint_actuator.h to force it to do so.

huihuaTRI, 2020-03-02:
I had this problem multiple times. @sherm1 pointed out that this information is already in Doxygen, but hidden in a drop-down link, for example, search for `Public Member Functions Inherited From ...`. If we can let it show by default, that would solve the problem mostly. 

The issue that you won't be able to search for such an API still persists though.

jwnimmer-tri, 2020-03-02:
You can try the fix locally -- add a `using` to the join actuator header then regenerate doxygen (`bazel-bin/doc/doxygen --quick multibody`).  See https://github.com/RobotLocomotion/drake/blob/d243bba621cb97d87bf351b9d0137e108ef84640/systems/framework/input_port.h#L166-L175 for an example of the using-statement.

huihuaTRI, 2020-03-02:
That works. The inherited member functions will still be hidden there, which is a duplicate if you click that drop-down menu. 

Would it still be nicer if just let Doxygen show these functions by default? Then we don't need to worry about adding `using` and missing new functions added in the parent class.

As for member function searching, the method you suggested also only shows the base class. . I guess this is not necessarily an issue though.

jwnimmer-tri, 2020-03-02:
> Would it still be nicer if just let Doxygen show these functions by default?

We tried flipping this default, but it makes things like the primitive systems unusable.  The docs for, e.g., https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_gain.html would show the bajillion LeafSystem methods so basically all of https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_leaf_system.html is pasted into _every_ system's Doxygen.

So the bottom line is that as best I know currently, authors should add using-statements to make the documentation suit their purpose.  (Or, avoid implementation inheritance.)

huihuaTRI, 2020-03-03:
Thanks for the info. FYI @sherm1

yu-fz, 2023-03-08:
I noticed that Drake MultibodyPlant has a function `Mbp::GetJointIndices(ModelInstanceIndex)` that returns a vector of all joint indices associated with the given model instance. 

Would it make sense to make a `GetActuatedJointIndices(...)` version of this method that will return a list of only the actuated joints associated with a model instance? Since there appears to be no convenience method for returning the list of all actuated joint names for a model instance, I think this method would make it much easier to set states of closed loop kinematic chains where only a subset of its joints are actuated. 

I also do not see anything in the `multibody::Joint` class documentation that allows users to query if a joint is actuated, which would permit me to loop over all regular joint indices and set states for the joints that have joint actuators attached.

amcastro-tri, 2023-03-08:
The usual way to know if a joint is actuated is to loop over JointActuators and if you need it, record the joint being actuated.
We can definitely add something like `GetActuatedJointIndices()` which would accomplish exactly that. Leh me open an issue.