RobotLocomotion/drake #12782: brew cask install drake

jamiesnape, 2020-02-27:
Relates #1183.

Work in progress:
```
brew tap jamiesnape/drake
brew cask install drake
```

jamiesnape, 2020-09-30:
`brew install --cask robotlocomotion/director/drake` is not available for the monthly releases. The cask does includes SNOPT support and installs both Homebrew and pip dependencies. Drake and pip dependencies are in `/opt/drake`, while Homebrew dependencies are in `/usr/local`.

jamiesnape, 2020-09-30:
Oh dear,
```
“_module_py.so” cannot be opened because the developer cannot be verified.

macOS cannot verify that this app is free from malware.

Homebrew Cask downloaded this file on September 19, 2020 from drake.mit.edu.
```

jamiesnape, 2020-09-30:
`brew install --cask --no-quarantine robotlocomotion/director/drake` is a workaround, but certainly not a desirable workaround.

jwnimmer-tri, 2022-02-15:
I am curious to hear from any macOS users who might want `cask install drake` as a C++ dependency?

My current thinking is that `pip install drake` (#15958) is much more important, and would meet the needs of many macOS users.  The option to download a [stable tar.gz archive](https://drake.mit.edu/from_binary.html#stable-releases) would remain for those who need C++.

Neotriple, 2022-03-12:
@jwnimmer-tri 

Think this would b

> I am curious to hear from any macOS users who might want `brew cask install drake` as a C++ dependency?
> 
> My current thinking is that `pip install drake` (#15958) is much more important, and would meet the needs of many macOS users. The option to download a [stable tar.gz archive](https://drake.mit.edu/from_binary.html#stable-releases) would remain for those who need C++.
> 
> I might close this ticket as "won't do anytime soon" if I don't hear any interest from users for `brew` support.

Would love to have this option.

jwnimmer-tri, 2022-10-11:
A lot of people have upvoted this since I last looked.  I no longer plan to close this as "won't fix".  We'll keep it on the roadmap.

As far as prioritization, I still plan on finishing the remaining `pip install` problem (#15954) first, before we dig in with `brew`.

jwnimmer-tri, 2024-03-16:
Having more experience with Homebrew over the past couple years, I don't think this will practical for us to accomplish.  Closing as "not planned".

For some users, the wheel builds via https://drake.mit.edu/pip.html will be a nice alternativel

Otherwise, the binaries we provide at https://drake.mit.edu/from_binary.html can be downloaded and unzipped to get a pre-compiled build of Drake that works on macOS, in a way that's compatible with Homebrew.

closed 2024-11-17 by jwnimmer-tri