RobotLocomotion/drake #12789: Use terms inboard/outboard instead of parent/child in internal MBT code.

amcastro-tri, 2020-02-28:
Per [review comment here](https://reviewable.io/reviews/robotlocomotion/drake/12764#-M11n_4AAPWLGXJvhUIh), the parent/child terminology might be somewhat fuzzy inside a context that already uses that terminology (for instance MBT parent of a body, or parent/child bodies in an SDF joint). 
Therefore it'd be a good idea to refactor this internal MBT code to use the inboard/outboard terminology, which is widely accepted in the community.

This could be a great first PR for Joe!

mitiguy, 2020-09-02:
The inboard/outboard terminology and related issues (#12789 and #13040) arose in PR# 13961.
it would be helpful to first agree/converge on frame names for documentation and notation.  
For example, we currently use Parent body P with frame F and Child body B with frame M.
Perhaps this should be Parent body P with frame Jp and Child body C with frame Jc.
For topology, perhaps we could use different words such as A/B or inboard/outboard or upstream/downstream or ...

amcastro-tri, 2020-09-03:
I've been consistently using P/F with B/M at the "mobilizer" level (internal tree model of our world) and P/Jp and B/Jc at the "joint" level (physical representation of our world).
If somewhere in Drake you find an inconsistency, it should be fixed. But I believe we did like this in the past, given that mobilizers will eventually be reversible in order to enable the modeling of arbitrary loop kinematics.

PS. I removed comments from #13040 since there is no need to have the same discussion sprinkled around. Besides, it belongs here.

sherm1, 2020-09-03:
I'll assign this one to @mitiguy for now since he's currently working on joint terminology docs.

jwnimmer-tri, 2024-11-01:
This is the kind of ticket where either we solve it right away, or should just close it as "not planned".  Maybe someone solved it in the past 4.5 years and didn't close this.  If not, then it doesn't matter anyway.

closed 2024-11-01 by jwnimmer-tri