RobotLocomotion/drake #12790: Document how to model a force/torque sensor in Drake

sherm1, 2020-02-28:
We have successfully modeled force/torque sensors by placing a Weld joint where the sensor is and using MBPlant's reaction force output port to obtain the force and torque at the Weld, which is what a force/torque sensor there would read.

Unfortunately that's a bit obscure. In #292 @RussTedrake wrote
> nice! might it be worth adding that idea to the documentation somehwere? 
> so it would come up on a search? (wouldn't it be nice if doxygen provided that search...)

In the absence of a more straightforward way to model these sensors, we should figure out some "obvious" place in the documentation where a user would find suggestions on how to model force sensors and perhaps other common model elements using Drake's feature set.

Any ideas where to put that?

(Assigning initially to @huihuaTRI since you have used this feature and might know where you would have looked.)

RussTedrake, 2020-02-28:
Perhaps adding to the doxygen here?
https://drake.mit.edu/doxygen_cxx/group__sensor__systems.html

RussTedrake, 2020-02-28:
and/or somewhere in here: https://drake.mit.edu/doxygen_cxx/group__multibody.html

PR #12791 (2020-03-02): Add doc about the availability of force/torque sensor
Add some documentation about how to use the `get_reaction_forces_output_port()` to model a force torque sensor.

Resolves #12790

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12791)
closed 2020-03-02 by huihuaTRI