RobotLocomotion/drake #12793: libClp.so (coin-or) missing from `/opt/drake/share/drake/setup/packages-bionic.txt`

RussTedrake, 2020-02-29:
I tried using only `/opt/drake/share/drake/setup/packages-bionic.txt` to provision colab in order to `import pydrake`, but it currently fails.

Running the following on colab.google.com:
```
try:
  import pydrake
  import underactuated
except ImportError:
  !curl -s https://raw.githubusercontent.com/RussTedrake/underactuated/colab_use_drake_binary_prereqs/scripts/setup/jupyter_setup.py > jupyter_setup.py
  from jupyter_setup import setup_underactuated
  setup_underactuated()

# Setup matplotlib backend (to notebook, if possible, or inline).  
from underactuated.jupyter import setup_matplotlib_backend
plt_is_interactive = setup_matplotlib_backend()
```
results in 
```
/opt/drake/lib/python3.6/site-packages/pydrake/common/__init__.py in <module>()
----> 1 from ._module_py import *

ImportError: libClp.so.1: cannot open shared object file: No such file or directory
```

The change in this branch is visible here:
https://github.com/RussTedrake/underactuated/pull/316

jamiesnape, 2020-03-02:
Should be there https://github.com/RobotLocomotion/drake/blob/c587248be03d450a9a5361e69bccb77ae6c6d3fd/setup/ubuntu/binary_distribution/packages-bionic.txt#L1

RussTedrake, 2020-03-03:
Resolved.  Jamie realized that I had not called apt-get update before installing the prereqs.  My apologies!

closed 2020-03-03 by RussTedrake