RobotLocomotion/drake #12807: Q size for floating bodies is unclear from docs

ggould-tri, 2020-03-03:
It is not clear from the documentation of `Body` how many positions the body has.  This makes it difficult to correctly use `Body::floating_positions_start()`.

I ran into this trying to pretty-print a positions vector; to annotate any qs related to floating bodies I had to write:
```
    for (BodyIndex b : plant.GetFloatingBaseBodies()) {
      const Body<double>& body = plant.get_body(b);
      if (body.is_floating()) {
        int num_dofs = body.has_quaternion_dofs() ? 7 : 6;
        int index_in_body = i - body.floating_positions_start();
        if (index_in_body >= 0 && index_in_body < num_dofs) {
          output << body.name() << "[" << index_in_body << "] == "
                 << pos[i] << std::endl;
          found = true;
          break;
        }
      }
    }
```
The 6-or-7 hardcoded constants there really should be accessed through a method on `Body` rather than through esoteric user knowledge.  Particularly as the 6 is not obviously correct -- it embodies an assumption about the underlying mobilizer.

sherm1, 2020-03-03:
Is there a better way to do this @amcastro-tri ?

amcastro-tri, 2020-03-05:
I'd like as much as possible a homogeneous API. That is, I honestly don't like what we have right now.
Bodies do have something in common, and that is that they all have mobilities assigned to them.

ggould-tri, 2020-03-05:
I think that would make a lot of sense.

(FYI, the example code above is in files I am trying to migrate to Drake eventually; you can now see it in all its horror at https://github.com/RobotLocomotion/drake/pull/12824 )

edrumwri, 2020-03-26:
@amcastro-tri I think you want position_start() for consistency with multibody::Joint.

amcastro-tri, 2020-03-26:
do you mean `Body::position_start()` @edrumwri? do you agree with my the proposal above?

edrumwri, 2020-03-26:
Yes, @amcastro-tri, except Body::position_start()- I otherwise agree with your proposal. Note that by deprecating Body::floating_positions_start(), you could address #12757 simultaneously.

jwnimmer-tri, 2024-03-16:
> I ran into this trying to pretty-print a positions vector; to annotate any qs related to floating bodies I had to write ...

Since the time the OP comment was posted, now in MbP floating bodies always have associated joints.  So, the way to print a labeled `q` vector is to use loop over all joints and use the `Joint::position_suffix` to get a marker name for a nice name on each `q[i]`.

Similarly, the way to know how many positions a Joint has is to ask the joint.

It doesn't not make sense to as a _body_ how many positions it has, so I'm closing this as "wont' fix".

closed 2024-03-16 by jwnimmer-tri