RobotLocomotion/drake #12810: Rename files for computing contact surfaces between [surface,volume][mesh, plane, half space].

DamrongGuoy, 2020-03-04:
Rename files for computing contact surfaces in `geometry/proximity/` like this:
```
mesh_intersection.h -> intersection_surface_volume.h
mesh_plane_intersection.h -> intersection_volume_plane.h
mesh_half_space_intersection.h -> intersection_surface_half_space.h
```
`intersection_surface_volume.h` is for a rigid surface mesh contacts a soft volume mesh with pressure field.

`intersection_volume_plane.h` is for a soft volume mesh with pressure field contacts a rigid plane.

`intersection_surface_half_space.h` is for a rigid surface mesh contacts a soft half space.

CC: @sherm1 , @amcastro-tri , @SeanCurtis-TRI , @tehbelinda

amcastro-tri, 2020-03-04:
Is the distinction between half space and plane that a plane is the boundary of a half space?

SeanCurtis-TRI, 2020-03-04:
@amcastro-tri. That *is* the origin of the plane. It's for contact with a rigid half space.

DamrongGuoy, 2020-03-04:
Yes, @amcastro-tri the plane is the boundary of a half space.  The plane has infinite area but no volume: {(x,y,z): Ax + By + Cz = D}.  The half space has infinite volume: {(x,y,z): Ax +By + Cz <= D}.

closed 2024-03-20 by jwnimmer-tri