RobotLocomotion/drake #12815: dot product with symbolic::Variables does not understand to convert to Expression.

RussTedrake, 2020-03-05:
The code:
```
  MathematicalProgram prog;
  const auto x = prog.NewIndeterminates(num_states, "x");
  std::cout << x.dot(x) << std::endl;
```
fails with 
```
external/eigen/include/_usr_include_eigen3/Eigen/src/Core/GenericPacketMath.h:171:37: error: could not convert 'drake::symbolic::operator*(drake::symbolic::Expression, const drake::symbolic::Expression&)(drake::symbolic::Expression((* & b)))' from 'drake::symbolic::Expression' to 'drake::symbolic::Variable'
         const Packet& b) { return a*b; }
```

It's not a blocker, as I can use `(x.transpose()*x)[0]` -- but it's less clean and the error above might confuse a novice.

Can we teach `Vector<Variable>.dot(Vector<Variable>)` to return an `Expression`?

soonho-tri, 2020-03-05:
https://github.com/RobotLocomotion/drake/blob/6952d8a4cf9381695ec0bc50c1df99f370437d3f/common/symbolic_expression.h#L1149-L1155

We teach Eigen the fact that any binary operation, `Variable bop Variable` should return an Expression. However this is not respected.

`v.dot(v)` is going to execute the following code. The return type `ResScalar` of `run` method is `Expression` (which is what we want).

https://github.com/eigenteam/eigen-git-mirror/blob/777b928d701702e2c8f8fa50643688678bdd727b/Eigen/src/Core/Dot.h#L29-L39
```c++
struct dot_nocheck
{
  typedef scalar_conj_product_op<typename traits<T>::Scalar,typename traits<U>::Scalar> conj_prod;
  typedef typename conj_prod::result_type ResScalar;
  EIGEN_DEVICE_FUNC
  EIGEN_STRONG_INLINE
  static ResScalar run(const MatrixBase<T>& a, const MatrixBase<U>& b)
  {
    return a.template binaryExpr<conj_prod>(b).sum();
  }
};
```

However, the `sum()` function below returns `Variable`, not `Expression` as its return type should be the `Scalar` of the input vector.

https://github.com/eigenteam/eigen-git-mirror/blob/777b928d701702e2c8f8fa50643688678bdd727b/Eigen/src/Core/Redux.h#L436-L443
```c++
template<typename Derived>
EIGEN_DEVICE_FUNC EIGEN_STRONG_INLINE typename internal::traits<Derived>::Scalar
DenseBase<Derived>::sum() const
{
  if(SizeAtCompileTime==0 || (SizeAtCompileTime==Dynamic && size()==0))
    return Scalar(0);
  return derived().redux(Eigen::internal::scalar_sum_op<Scalar,Scalar>());
}
```

This is the main reason why the code does not compile.

I'll check what we can do and keep you posted here.

jwnimmer-tri, 2021-11-18:
This still exists with Eigen 3.4.  I would say the problem here is that Eigen's `conj_helper` is disobeying `ScalarBinaryOpTraits`, and that the best response is to submit an issue or pull request upstream, and not try to hack around it in Drake.

@soonhokong are you interesting in writing up that ticket?  Otherwise, I'll come back to this and file it a bit down the road.

closed 2024-03-16 by jwnimmer-tri