RobotLocomotion/drake #12817: Need some sugar to test if two polynomial symbolic::Expressions have coefficients that are approximately equal.

RussTedrake, 2020-03-05:
Right now I'm using:
```
  EXPECT_TRUE(Polynomial(V - V_expected)
                  .RemoveTermsWithSmallCoefficients(1e-6)
                  .EqualToAfterExpansion(Polynomial(Expression::Zero())));
```
It's not horrible, but took me a while to figure out a version I liked -- partly because I kept forgetting that `Expression::ToPolynomial()` returns a `Polynomiald` and NOT a `symbolic::Polynomial`.

hongkai-dai, 2020-03-11:
@soonho-tri , have you worked on this already? I can take this one.

soonho-tri, 2020-03-11:
@hongkai-dai , I couldn't start working on this yet. It would be great if you can take this. Thank you.

hongkai-dai, 2020-03-11:
I plan to add a new function to `symbolic_polynomial.h`
```
bool symbolic::Polynomial::AlmostEqual(const symbolic::Polynomial& other, double tolerance);
```
which returns true if the coefficients between two polynomials are all equal up to the tolerance.

RussTedrake, 2020-03-11:
Thanks!  I think something like `CoefficientsAlmostEqual` is more clear?

PR #12853 (2020-03-12): Add Polynomial::CoefficientsAlmostEqual function.
A syntax sugar to compare if two polynomials are almost equal.

Resolves #12817

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12853)
closed 2020-03-12 by ggould-tri