RobotLocomotion/drake #12833: cannot set/maintain decision_vars vs indeterminates when manipulating symbolic::Polynomial

RussTedrake, 2020-03-08:
As a result, in #12827, I have to go from Polynomial back to Expression and back again, so that mathematical program knows which is which.

```
  const auto rho = prog.NewContinuousVariables<1>("rho")[0];
  std::cout << V.indeterminates() << std::endl;
  std::cout << (V-Polynomial(rho)).indeterminates() << std::endl;
```
yields
```
{x(0), x(1)}
{x(0), x(1), rho(0)}
```
even though `rho` is a decision variable.  As a result I have to write:
```
  prog.AddSosConstraint(((V - Polynomial(rho)) *
                        Polynomial(pow((x.transpose() * x)[0], d)) -
                        lambda * Vdot).Expression());
```
which is both ugly and inefficient.

hongkai-dai, 2020-03-08:
I think the problem is when we write `Polynomial(rho)`, it defaults to treat all variables as indeterminates. You will have to provide the list of indeterminates when constructing the polynomial in this case. Here is the code that works for me
```C++
const auto rho = prog.NewContinuousVariables<1>("rho")[0];
std::cout << V.indeterminates() << std::endl;
std::cout << (V-Polynomial(rho, {})).indeterminates() << std::endl; // This constructs Polynomial rho with empty indeterminates.
```

RussTedrake, 2020-03-08:
I would have to pollute every line with things like this.  It think it's a sign of something more broken about the relationship between `MathematicalProgram` and `symbolic::Polynomial`.  I know you and @soonho-tri debated about this when you were first bringing things online... but from today's experience, I would think that:

- `Expression`s do not keep a list of decision variables / indeterminates.  I don't think `Polynomial`s should either.
- `MathematicalProgram` is the only class that has to keep these two straight.  In the same way that `Expression`'s get parsed into decision variables/ indeterminates when they are passed in to e.g. `AddSosConstraint`, `Polynomial`s should have that, too.

hongkai-dai, 2020-03-08:
If Polynomial doesn't know which variable is indeterminate, and which variable is polynomial, then it is hard to know whether an expression is a polynomial or not. For example, the expression sin(a) * x^2 is not a polynomial if both a and x are indeterminates; it is a polynomial if only x is indeterminates.

RussTedrake, 2020-03-09:
Does `symbolic::Polynomial` really support `sin(a) * x^2`?  Is there a case where it is meaningful to call that a `Polynomial`?

Even if yes, I would say that it could be ok for a user to have to deal with labeling indeterminates when creating Polynomials for cases like that.  But I would think that, almost independent of what the Polynomial class lists as an indeterminate, the `AddSosConstraint(Polynomial)` method should use MP's list of indeterminates (only) as the indeterminates, MP's decision variables as decision variables, and throw if there are any other Variables in the Polynomial?

soonho-tri, 2020-03-09:
@RussTedrake , FYI, we already have the following method:

```c++
   /**
    * Adds constraints that a given symbolic expression @p e is a sums-of-squares
    * (SOS), that is, @p e can be decomposed into `mᵀQm`. Note that it decomposes
    * @p e into a polynomial with respect to `indeterminates()` in this
    * mathematical program. It returns a pair expressing:
    *
    *  - The coefficients matrix Q, which is positive semidefinite.
    *  - The monomial basis m.
    */
   std::pair<MatrixXDecisionVariable, VectorX<symbolic::Monomial>>
   AddSosConstraint(const symbolic::Expression& e);
```

I'd like to emphasize the note here:

> Note that it decomposes @p e into a polynomial with respect to `indeterminates()` in this mathematical program.

soonho-tri, 2020-03-09:
Some thoughts:

 1. We can add a sugar, `symbolic::Polynomial MP::MakePolynomial(Expression)` which creates a polynomial using the MP's indeterminates.
 2. Then we can provide overload for the methods which currently only take a `symbolic::Polynomial` to take a `symbolic::Expression`. Internally, they create a polynomial using `MakePolynomial` in 1) then call the existing methods. `AddSosConstraint` above is an existing example.

hongkai-dai, 2020-03-09:
@soonho-tri Russ brought up this issue because we want to avoid calling `MP::AddSosConstraint(Expression e)`. The reason is that we have two polynomials `p1, p2`, `p1` is already a symbolic::Polynomial object, `p2` is a simple expression. and we want to enforce that `p1 + p2` is a SOS polynomial. Currently we have two possible workflows
1. Create a symbolic::Expressions e=p1 + p2, and then call `AddSosConstraint(e)`. I argued that this involves converting `p1 + p2` to `e`, and then converting `e` back to a polynomial. The conversion is non-trivial, and could be time consuming for complicated polynomials.
2. When constructing the symbolic polynomial `p2`, provide the list of indeterminates as Polynomial p2(p2_expression, indeterminates), then call `AddSosConstraint(p1 + p2)`. In this way we don't need to convert the sum `p1 + p2` from a symbolic expression to a symbolic polynomial; we only need to parse the simple expression `p2` as a polynomial. So computationally this is easier than approach 1. The caveat is that the user must know the indeterminates when constructing Polynomial p2. This puts extra burden on the user to remember the indeterminates, while `MathematicalProgram` knows the indeterminates.

soonho-tri, 2020-03-09:
@hongkai-dai , thanks for the clarification. Here are some options that I can think of:

1. Add `operator+(Polynomial p, Expression e)` and internally we construct a polynomial for `e` using `p.indeterminates`. Previously, I wanted to avoid this one since a user might not understand what's happening under the hood.

2. Another option is to write `AddSosConstraint(p + MP::MakePolynomial(e))`. This is essentially the same thing as your option 2, but with a sugar `MP::MakePolynomial`, a user doesn't have to explicitly pass the indeterminates.

hongkai-dai, 2020-03-09:
@soonho-tri Thanks! I like option 2 better. It doesn't involve too much change on symbolic::Polynomial class. So symbolic::Polynomial still knows the difference between indeterminates and decision variables.

@RussTedrake what do you think about option 2? Currently we do support sin(a) * x^2 to be a polynomial. We don't have a good use case yet. I agree it is tempting to make Polynomial class not to differentiate between indeterminates and decision variables (like what spotless does), but I am afraid that would require completely rewriting the whole class.

RussTedrake, 2020-03-09:
Regarding @soonho-tri 's post:
```
Note that it decomposes @p e into a polynomial with respect to indeterminates() in this mathematical program.
```
I think that may not be true?  Perhaps it's a bug?  But I definitely got different results when I called `AddSosConstraint(Polynomial)` vs `AddSosConstraint(Polynomial(Expression))` which were resolved when I changed it to `AddSosConstraint(Polynomial(Expression, prog.indeterminates()))`.

soonho-tri, 2020-03-09:
@RussTedrake , FYI, the following should give us the same result:
 1. `mp.AddSosConstraint(e)`
 2. `mp.AddSosConstraint(Polynomial(e, mp.indeterminates()))`

However, the following is different:
 3. `mp.AddSosConstraint(Polynomial(e))`

hongkai-dai, 2020-03-09:
Another option is to add a new type to symbolic::Variable as `Indeterminates` in https://github.com/RobotLocomotion/drake/blob/7875b8b44cf0bd373ccad8b761956c57055cd3f4/common/symbolic_variable.h#L39-L50

Then we we call `NewIndeterminates`, the returned variables will have type `Indeterminate`. When we construct Polynomials, the constructor automatically knows whether a variable is an indeterminate or a decision variable.

hongkai-dai, 2020-03-09:
Following our discussion on why using symbolic::Polynomial can be faster than using symbolic::Expression (and then convert the expression to polynomial in AddSosConstraint):

Suppose that we want to verify the following two conditions:
```
p1 - p2 is SOS
p1 - p3 is SOS
```
note that `p1` appears in both conditions.
Suppose we are given
```
symbolic::Expression e1 = p1;
symbolic::Expression e2 = p2;
```
If we call `mp.AddSosConstraint(e1 - e2)` and `mp.AddSosConstraint(e1 - e3)`, we need to convert both `e1 - e2` and `e1 - e3` to symbolic::Polynomial. On the other hand, if we use symbolic Polynomial
```
symbolic::Polynomial p1 = e1;
symbolic::Polynomial p2 = e2;
symbolic::Polynomial p3 = e3;
mp.AddSosConstraint(p1 - p2);
mp.AddSosConstraint(p1 - p3);
```
the conversion takes less time, as we don't need to convert `e1` to Polynomial in both `e1 - e2` and `e1 - e3`. This is especially true when `e1` is complicated, or `e1` is a symbolic::Polynomial to start with (for example, it is returned from mp.NewSosPolynomial()).

soonho-tri, 2020-03-09:
@hongkai-dai , I see. I agree that in this case creating polynomials for subexpressions works best.

With the mentioned sugar `MP::MakePolynomial`, we can re-write the example as follows.

```c++
symbolic::Polynomial p1 = mp.MakePolynomial(e1);
symbolic::Polynomial p2 = mp.MakePolynomial(e2);
symbolic::Polynomial p3 = mp.MakePolynomial(e3);
mp.AddSosConstraint(p1 - p2);
mp.AddSosConstraint(p1 - p3);
```

Let me prepare a PR for that. Of course, we can keep discussing and find a better solution.

hongkai-dai, 2020-03-09:
Thanks Soonho.

I think adding `mp.MakePolynomial` should work. It requires the user to call this function. On the other hand, I think adding `Indeterminate` type makes the code easier to use symbolic::Polynomial p1(e1)` directly. Do you concern that adding Indeterminate type will cause ramification in the codebase?

RussTedrake, 2020-03-10:
I agree that `mp.MakePolynomial` could work, but still wish the user didn't have to think about it.  And, in terms of performance as discussed above, would we also offer `mp.MakePolynomial(Polynomial)`... perhaps differently named, that takes a polynomial and rejiggers which variables are the indeterminates?

soonho-tri, 2020-03-10:
> I think adding Indeterminate type makes the code easier to use symbolic::Polynomial p1(e1)` directly. Do you concern that adding Indeterminate type will cause ramification in the codebase?

1) The `type` field of a `symbolic::Variable` determines the range of values that it can hold. For example, `CONTINUOUS` indicates that it can take a `double` value, `INTEGER` indicates it can take an `int` value, etc. The distinction between a decision variable and an intermediate is orthogonal to this range information. So, I think we cannot extend the existing `type` field. We need to introduce a new field to it.

2) "Is this information intrinsic to this class?" is the question that I ask myself when I consider any extension to an existing class. In this case, I think the distinction between decision variables and indeterminates is only meaningful to the MathematicalProgram while we use `symbolic::Variable` other places. So I try to provide an alternative solution.

hongkai-dai, 2020-03-11:
> 1. The `type` field of a `symbolic::Variable` determines the range of values that it can hold. For example, `CONTINUOUS` indicates that it can take a `double` value, `INTEGER` indicates it can take [...]

That is a compelling argument. I am convinced that we should not add indeterminate field to symbolic::Variable. Thanks for the explanation.

hongkai-dai, 2020-05-14:
This is resolved in #12847

closed 2020-05-14 by hongkai-dai